opentx/opentx #8010: Can not flash the firmware to the folder "G" of the Horus X12S

nscch, 2020-10-09:
Hi

I got a problem with the Horus X12s. I can not flash the Firmware form 2.3.9 to the new firmware 2.3.10 from the Companion to the Horus X12S. 
The  following message appears:
 "Kann nicht speichern G//FIRMWARE.BIN (Grund: Unbekannter Fehler)"

Have you got an idea, why does it not work?

Thank you very much for your ideas.

Best Regards

Norbert Schlegel

kilrah, 2020-10-09:
You must be in bootloader mode to flash firmware.

closed 2020-10-09 by kilrah

nscch, 2020-10-10:
Hi Andre

Yes, but when I plug the USB with the rtansmitter turn off, there is
nothing comes on.
I have attempted with direct boot over the DfuSe-Tool, but it doesn' t work.

Is it possible, that the boatloader of the Horus is not working?

Best Regards

Norbert

kilrah, 2020-10-10:
Bootloader is accessed by turning on the radio while holding both horizontal trims inwards.

nscch, 2020-10-11:
Hi Andre

Your advice was successful. Thank you very much!

Now I got a really big problem with my Taranis X9D (not plus).
I connected the transmitter to my computer (Win 10) in boot mode for the
update from 2.3.9 to 2.3.10 over companion.
After a moment, the boot screen of the Tx is totally black. The message in
computer appeared, that the USB-Device is broke down or damaged.
Since this event, I could not switch the transmitter anymore. There is
nothing to here and the screen is even still black.
When I connect the Taranis to my computer now, it couldn't find the
stm-device of the transmitter no more.
So, I could imagine, that one of the boards inside the taranis was damaged.
The main switch or even the backboard with the USB connector inside, or...

Have you a good idea?

Thank you very much for your response.

Best Regards

Norbert