opentx/opentx #8016: Jumper T18 "No SD card" error after flashing 2.3.10

HannoKranzhoff, 2020-10-10:
When I flash my Jumper T18 with version 2.3.10 and write the SD card content 
the TX fails to boot and just displays "No SD card" (white text on black screen).

After reverting to the original firmware the TX works as expected again.

**To Reproduce**
Steps to reproduce the behavior:

- flash 2.3.10 on Jumper T18
- write SD content for T18 onm SD card
- reboot

**General information**
- opentx-t18-lua-noheli-en version 2.3.10
- Hardware Jumper T18
- OpenTX Companion 2.3.10 stable, Oct. 6, 2020

3djc, 2020-10-10:
This is a know hardware defect of Jumper T18, hidden by factory installed firmware. Please contact your radio dealer, they should have received instructions from Jumper on how to fix this

closed 2020-10-10 by 3djc

HannoKranzhoff, 2020-10-10:
Ok, thanks a bunch. Didn't know that.

HannoKranzhoff, 2020-10-12:
Just got a reply from Jumper:
"You can download firmware here https://www.jumper-rc.com/download/t18-series/
Please update firmware to the one on our website."

The version available for download on their website is 2.3.8. :(

Does this mean I will always have to wait for Jumper to release new firmware versions?
What kind of hardware issue causes this issue?

kilrah, 2020-10-12:
Bad soldering on the SD card holder. They were sending replacement motherboards to people with the issue initially but I guess they got tired of the expense caused by their poor QC... try your dealer instead.