opentx/opentx #8025: 2.3.10+ can no longer overide preference to turn backlight ON via global function w/ Action "Backlight"

pitts-mo, 2020-10-12:
After upgrade to 2.3.10+ I can no longer turn back light ON or keep it ON via global function w/ Action "Backlight".

Is there a new method to allow this behavior or should I investigate this as a potential bug?

More details:
I have seen this behavior on x9d+ and tx16s. 
I utilize on keys set to turn back off after 5 seconds.
My historical settings show I utilized a global function with:
Switch SF↓,
Action Backlight,
Parameters *None*, 
Enable not ticked.

With 2.3.10 and 2.3.11 build from git today I can not find any way to get my global function for backlight to turn on and stay on. The Backlight is only observed to come on with key usage which will time out based on the preference I have set.

kilrah, 2020-10-12:
Changed in https://github.com/opentx/opentx/pull/8018

closed 2020-10-12 by kilrah

pitts-mo, 2020-10-12:
Yes, building from the 3djc/improve-backlight pull request for #8018 does restore the functionality I have become accustomed to. 
Thank you @3djc and @kilrah 
-p