opentx/opentx #8026: Spektrum Telemetry

whasianben, 2020-10-12:
I have a Radiomaster TX16S that I have bound to a Spektrum 4649T receiver. I have tried to set up telemetry on it so I can get a voltage and current draw reading, but when I went to discover new sensors there were a ton of vague and misleading sensors. I was only able to use the voltage reading accurately but everything else for the most part was useless. Would it be at all possible to better adapt OpenTX to support Spektrum telemetry in this manner? It would also be nice to be able to use the telemetry on this radio to change Betaflight settings like VTX channel and PIDs.

Dosh22, 2021-03-26:
I have the same problem with Spektrum 4650 receiver.

2pacsawyer, 2021-09-08:
Im also having the same issue with a Spektrum 4650. All I really care about is RSSI.

3djc, 2021-09-09:
Telemetry is handled by multimodule, and as far as I know, it forward most (all) sensors. Anyhow, this is more for multimodule than otx

closed 2021-09-09 by 3djc