opentx/opentx #8027: [AVR model] Companion 2.1 seems to be ignoring the "frsky" option in firmware download

gkvirkvia, 2020-10-13:
Hi folks,

I recently dusted off my Turnigy 9x with DHT telemetry mod done on it. It had OpenTX 2.2 on it and after seeing the eeprom corruption KI I downgraded down to 2.1.
This is where my issue showed up - it seems like the "frsky" version of the firmware is still reading the THR and AIL switched from the old ports. They are shown as always om when my drone is OFF and are flickering when the drone is ON and transmitting telemetry information.

I quadruple-checked that I'm flashing the correct firmware on it. My radio is Turnigy 9x with stock mega128 chip and DHT telemetry mod done. The file name I'm downloading(via companion2.1) and installing(via eePe) on it is opentx-9x128-frsky-en.hex

This is also not an issue of me messing up the hardware mod. The radio used to work great with OpenTX 2.2 on it. I also just flashed er9x on it, enabled the frsky mod option in the hardware menu and the switches work great. I also get telemetry info from my drone on 9x (can provide photos)

bigtwin, 2021-02-11:
I guess nobody would try to recreate or troubleshoot this as your software is way obsoleted. Why dont you try it with the lates OTX?

gkvirkvia, 2021-12-22:
@bigtwin because my TX is based on an AVR chip and newer software versions dropped support for those. OpenTX 2.1 is the only version I can use. It used to work, now its broken.

thijses, 2022-07-28:
i have exactly the same problem. I'm going to see if i can compile openTX myself, and manually fix this (assuming i'm capable).
@gkvirkvia did you ever manage to solve this already? (if so then i can avoid having to uncover the mysteries of openTX code)

gkvirkvia, 2022-07-28:
Nope, didn't invest enough time into this.

thijses, 2022-07-29:
compiling has been a nightmare (as expected). I'd rather not have to reinstall windows after replacing python 3.10 with 2.7, so i'm giving up (for now). I just installed ER9x on one of my transmitters and openTX (without telemetry) on the other, and i manually setup the same profiles on both. @gkvirkvia thank you for the speedy reply