opentx/opentx #8031: LUA improvement #7878 killed existing FM support on mixers in 2.3.10

lshems, 2020-10-13:
In above pull someone found it funny to change the behaviour of the getmix and setmix function. These functions where completely fine in  their way of working to support FM settings on mixes. It returned a byte for the FM that contained a well described set of bits for each flghtmode this mix should be active for (or not). So an EXSTING function is now NO LONGER WORKING as before, without any proper documentation on how it works now.

On top of that, this change wrongly implies that it would have fixed a request I made to allow access from LUA to the Flightmode switch assignments: so on which switch needs a flightmode to get active, as this function is MISSING.

I do understand that mistakes are made. But asking to fix apparent BUGS in existing functions always gets answered with a NO, it is already implemented, we dont want to change it.

And now you change a correctly working functionality without even documenting it, and any clue you might get from the request being referred to is something completely different from that what was changed.

Sorry, but I don't get the way you are working anymore. I really do not understand how you can think of people keep on taking the effort to develop scripts, if these things happen.

By the way, all LUA documentation seems to be gone. Perhaps it is now hosted somewhere else, but it wont help either.

A not so happy user of OpenTX, as if it matters.

3djc, 2020-10-13:
Duplicate #8021

closed 2020-10-13 by 3djc

lshems, 2020-10-13:
This is a completely different issue! Please read.