opentx/opentx #8032: 2.3.10 lua getMix returns completely unusable info about active flightmodes in an array with unusable keys

lshems, 2020-10-13:
The FM parameter in the LUA getMix function seems to return key-value pairs that shift when the active flightmodes are changed, making it a completely useless new functionality.

With all flight modes active on the mix, we get an array with key-value pairs 1-1, 2-2, up til 8-8 and then 0-0.
when deactivating flight modes 1 and 3 we get

1: 2
2: 4
3: 5
4: 6
5: 7
6: 8
0: 0

What on earth is the logic behind this?

Either revert to the fully functional way it was working, with a very compact and efficient way of storing the active flightmodes in the bits of the FM param.
Or set the keys corresponding to the flightmode that is active to true, or 1.

And please make sure that a read of the parameters can be used without reprocessing to write it back again, and prevent a shift on keys as was done on the curve functions.

Thanks.

lshems, 2020-10-13:
Sorry, forgot, please don't use true as an indicator, unless you also use false. do not use null to indicate false.

If you would like to copy from one mix to another, any key-value pair with nill as value, is considered not exsting by lua. Copying from a mix line with all inactive to one with all active wouldn't change anything in that case.

Storchy1, 2020-10-14:
See my item [https://github.com/opentx/opentx/issues/8021] there is a bug in model.insertMix(...)

closed 2021-09-04 by stale[bot]