opentx/opentx #8039: Improve flexi theme (no hard-coded colors, more options)

tri4ng1e, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
I really like the idea of flexi theme. But it's not as flexible as I want it to be. There are two things that made me switch back to DarkBlue:

1. Some colors are hard-coded and cannot be changed, so I should stick to these colors to create a consistent setup.
2. Flexi theme has 5 colors to customize and it's really nice in comparison to what you can tweak in the default theme, but there are some scenarios where it's not enough. For example, I really want to have completely black theme (with black bg and top bar), but top bar and menu selection colors are bound to one "main" color, which is kind of unusable when both set to black.

**Describe the solution you'd like**
I looked in the code briefly and found that for now the maximum number of theme parameters is 5. If there is a possibility to increase that number to like 7 or 8, we could make the flexi theme way more flexible. I tried to increase that number up to 7 and built the firmware & companion, and it seems to be working fine in the simulator. `RadioData` parameter changed from 881u to 897u. I don't think this is huge change but I may not know some details underneath.

**Describe alternatives you've considered**
If we can't change the number of theme parameters, we can rearrange them so that all the colors in the theme will be calculated from the them. There are some drawbacks, though -- we should make an algorithm that will decide the brightness of a color and then apply a darker of lighter modifier to support dark and light themes.

Alternatively, I should be able to create a full-black theme without any settings instead of tweaking flexi.

**Additional context**
I prepared a build system on my PC. I have some skills in C++, so I can try to make it myself. But I am interested to get your opinion about how to solve this issue. And maybe, move it to another theme to keep people who already using it from changes.

3djc, 2020-10-15:
Having more than 5 choices would mean EEprom version changes so has to wait 2.4

tri4ng1e, 2020-10-15:
There is no problem, though. May I start, then?

tri4ng1e, 2020-10-17:
Can we use a file (`ini` or `json`) in the theme's folder instead of settings menu? For one special theme, for example. To eliminate "new SD card version shipped and settings's file was overwritten" problem we can use the approach that is popular in Linux: to ship a file with default settings (or just an example), and to use another file (which user will be editing) to override the default values. In that way we can map all theme's colors without upgrading the EEPROM version or even increasing it's size. We can even do default and custom presets for that theme and choose them from the settings!

I was fiddling around black theme and tried to create a separate one with hard-coded colors. it has some rough pieces but already looks good to me. Some additional screens can be found [here](https://imgur.com/a/vTGgseT).

[image]

3djc, 2020-10-18:
No, that wont be allowed, we already had that discussion on another topîc and rules this option out.

What can be done however is allow lua to  read and write those colors

PR #8063 (2020-10-30): Add getColor LUA function
This closes #8028 
This closes #8039
closed 2020-10-30 by 3djc