opentx/opentx #8040: 2.3.10 TX16S: Active point does not follow curve in Input editor

RC-SOAR, 2020-10-15:
If an input is specified with two or more lines (e.g. for dual rates), then the graphic is not displayed correctly for the second line if it's inactive.

**To Reproduce**
Download [minimalist setup](https://www.dropbox.com/s/bb6w9oeaejpy9u5/GitHub_CurveBug.otx?dl=0).
Switch SA up to activate the second input line of input 'Ew'
Open the input editor on the second '74%' line of input 'Ew'.
Rotate S1

**Expected behavior**
The active point should follow the curve. It does not do so.
It works as expected in 2.3.9.
[image]
[image]

**Screenshots**
Please see screenshot

**General information**
 - OpenTX version (e.g. 2.3.10 nightly, but also reported on 3.1.10 stable version)
 - TX16S
 - OpenTX Companion 2.3.10 nightly

Hobby4life, 2020-10-15:
No need for a complex setup..

Just create a curve and see how ALL inputs follow CV1

[image]
[image]

kilrah, 2020-10-15:
Already mentioned in related PR for authors to comment. https://github.com/opentx/opentx/pull/7707

bsongis, 2020-10-16:
oups :)

lthole, 2020-10-20:
Reproduced on X10S 2.3.10

closed 2020-10-24 by 3djc

Helifunster, 2020-10-29:
Why is this issue closed?

3djc, 2020-10-29:
Because it has been fixed in nightlies

ahmad-kassem, 2020-10-29:
What do you mean fixed in nightlies?  How can we fix on the radio?

Helifunster, 2020-10-29:
Thank you. I think it was some spooky thing accordinly to Halloween season.

Regards

Helifunster, 2020-10-29:
Keep the superb job you are doing with OpenTX...but, the fastest, the better.

Regards

KTM-250, 2021-01-17:
Hello, I too am having the issue of input stick movement following CV1 throttle curve.  I'm totally new here so I can't make sense of these comments, can someone please explain how to fix this issue? Thanks in advance...

kilrah, 2021-01-17:
Update to OpenTX 2.3.11 that came out a week ago.

KTM-250, 2021-01-17:
Wow, very cool, I was ready to set this thing on fire... Thank you for
responding!!

PR #8057 (2020-10-24): Fix curves display on colorlcd
This fixes #8040