opentx/opentx #8042: Feature to turn on the radio in quiet mode (via switch)

tri4ng1e, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
I fly with different sound notifications and warnings, it's quite a powerful system with all the logical switches and stuff. But sometimes I use the radio as a joystick for the simulators (or just connect it to PC to sync SD and settings) and often in the evening and night time. I don't want to wake my wife up, though. I can turn the sounds off in the settings and create a "silent" model special for sim (which I already did), but I also have a startup and shutdown sounds which I really like while flying normally. 

**Describe the solution you'd like**
It's not a big problem, but I suggest to add a feature to enable quiet mode at the radio startup, toggled via switch (for example, trainer SH-down while turning on the radio) or other thing. Add a warning screen that radio is turned on in quiet mode for those moments it is triggered accidentally, not showing up if the radio set up to be quiet already through settings menu. Revert the settings back while turning the radio off if needed.

**Additional context**
I can try to do this myself.

aeronaut999, 2020-10-16:
That would be a nice feature.

kf4mot, 2020-10-19:
I would use this.

closed 2021-09-04 by stale[bot]