opentx/opentx #8046: X9D+ 2019SE OTX2.3.10 FrSky Tools SBEC Lua Error

pau33, 2020-10-17:
If I want to start Tools:SBEC, I get an Lua NiL comparison error.

You can see it in the TX but also in the Simulator.
A fix should not be too complicated.

All other toots at least start normally.

Hobby4life, 2020-10-20:
When running the Frsky SBEC script, I get a warning message on line 266.

[image]

Please make a screenshot of your error.

More info is needed.

pau33, 2020-10-20:
I get this message:
[image]

also the new neuron80 is connected (single SBEC the same):
[image]

I addressed this also in the Frsky firmware forum, as it might  be an Frsky problem (I hope it is not related to the X9D).

kreidler, 2020-12-21:
@pau33 Have you ever tried to delete the sensors and started a new search? Afterwards try LUA script again.

pau33, 2020-12-22:
Thank You for the tip.
It work's. With a completely new model the  lua script for the neuron works.
With the exiting model it doesn't. Although all values form the neuron are visible in the telemetry and it's id seams not to interfere.