opentx/opentx #8047: Companion: no "SP Power" flag for QX7

Tiefflieger68, 2020-10-17:
in Companion there is no flag for "SP Power".
In the System Settings of the Radio it is available and works well.

Tiefflieger68, 2020-10-17:
depends to QX7 Radios

kilrah, 2020-10-17:
What radios have it? I don't see it for any of them.

Tiefflieger68, 2020-10-18:
no Radio have this funktion in Companion.
This feature is imho only available Radios with external S.Port Connector. 
You are right, it is also not available in Companion for X10 Radio.

So this issue depends to all Radios with external S.Port Connector.

If I transfer the config of my QX7 to Conmpanion and back to the Radio, the Flag for "SP Power" is set to default (Auto).
So it would be great to have this funktion in Companion too.

commit bcc5f90c (2020-10-30): [companion] Add sport power support (fix #8047) (#8050)
closed 2020-10-30 by bsongis