opentx/opentx #8053: MacOS Companion 2.3.10 - Curve labels missing

lthole, 2020-10-22:
**Describe the bug**
In MacOS Companion 2.3.10, the Curve number labels are missing. The coloured blocks are there, but no text labels 'Curve 1', 'Curve 2', etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Companion 2.3 in MacOS
2. Go to the Curves tab, and look for the curve identification labels

**Expected behavior**
Was expecting to see curve identification labels.

**Screenshots**
See below

**General information**
 - OpenTX version: 2.3.10
 - Hardware: n/a
 - OpenTX Companion (2.3.10 on MacOS Catalina on MacBook Pro 15"
[image]

[image]

)

lthole, 2021-01-23:
The curve name labels are actually there, but they are the same colour as the background tiles. I found this by clicking on the Curves tab, and then changing the focus by clicking on the model in the .otx window.

bigtwin, 2021-02-11:
I can confirm - this is still the same in 2.3.11 (on MacOS Mojave).
Text and background are simply the same color.

closed 2022-04-17 by stale[bot]