opentx/opentx #8054: X12S - Mute

bsongis, 2020-10-22:
Hi Bertrand

Sorry to bother you , I just didn’t know really how to proceed , maybe you have some time to read through my “problem”.

I saw that you are very actively developing opentx and because I struggeled with something I came across a github bug report: [X10] mute audio when fifo buffer is empty #5875 (Link).

I have the same kind of problem with my X12s – whenever I have some external radio module plugged I get very noisy sound on my speaker.

I couldn’t find any bugfix except that link I mentioned before; but there is stated it was just tested on an X10 only – I would be very willing to support there with my X12 too … I just don’t know how.

Is there a way to get some support ? Can I also somehow participate to improve on opentx ?

I hope you find some time / mood to have a look at my “problem”.
 

Thanks a lot and sorry for bothering you.

BR
Stefan

bsongis, 2020-10-22:
There is a MUTE Pin on several radios, including X10, but I am not sure there is one on the Horus X12S

brainbubblersbest, 2020-10-28:
I can maybe help with measurement. Havin the same issue and are also interested, when this can be a possible solution. I use my x12 with almost no volume because off that.

brainbubblersbest, 2020-11-01:
Okay. Seems the Mute pin is soldered to ground. The only way seems to shutdown the amplifier via the sd pin. This one is traced to pi09.
But I dont know if that way would working good. 

Maybe Fros involve that vs1053a in this.

brainbubblersbest, 2020-11-02:
Now ill be 100% shure it's direct connected to ground and the other Marked chopper traces. So no Modification is possible and this must be handled in an other way by fros. 
[image]

brainbubblersbest, 2020-12-15:
Found something that Shows another x10 based Radio uses the SD Pin for This Mute Function. 
So maybe this can work. I would test this, if u utilize it this way in a test bin.
With crsf and r9m
 [image]

poge-st, 2020-12-27:
I am very willing to help on this topic as well - i can't help programming but if you need some support with testing i would be very happy to help - i have a crossfire module and a X12 - and this precise problem ;)

poge-st, 2021-03-17:
anyone got any updates on this topic ? :/

FLYN-001, 2021-05-15:
I have the same radio & same issue, I'd very much enjoy helping tp test a fix to this very distracting issue.

brainbubblersbest, 2021-05-15:
Please bertrand. We All would love this fix. I think it would be easy to Do that. The biggest Problem why i dont try this by myself is the Handy ap with compiling otx.
This Sound issue is very distracting 🙂

FLYN-001, 2021-05-20:
Yes please please please can we get a fix for this, a donation for fixing this would definitely be forthcoming!

brainbubblersbest, 2021-06-21:
1H:31M

FLYN-001, 2021-06-21:
My post from a Facebook group....

Ok so you got an X12S that buzzes like a mo’fo when you plug in your R9 or CRSF. Or is even audible with built in 2.4

Here’s a super neat trick to reduce that, but you do need eyes like a hawk, a steady blade & nerves of steel!

Remove the screen ribbon cable & cut traces 8 & 9 from the lower left side (backing side up). It’s easier to do it when it’s contacts up so lower right side when flipped over. 

These traces are for the headphone jack that is right behind your antennas. 

This circuit picks up loads of rf noise from your antennas & dumps it via a direct connection to your audio amp chip on the main board. 

Cut that connection & you can reduce the buzzing noise by 60-70%, the difference is very noticeable. However it will probably not stop it completely as the main board traces still run directly under the rf modules.

Also while the radio is open, extend the speaker cables a little & install a ferrite ring between them & that’ll be good for another nice reduction!

Plus, a small SMA extension cable for your antenna is another helpful thing!

It’s still not silent as the grave but it’s a damn-sight better than standard. I’ve done all of the above to my own radio. It all works. 

Disclaimer. 
If you do this it’s totally down to you, you MUST use a good magnifying glass or digital screen to enlarge & see what you’re doing. You MUST ONLY CUT 8 & 9. If you get it wrong & blow your radio up it’s on you. Not my fault if you get it wrong. 

Best thing to do is order a spare cable online & cut that one & test it is modified correctly BEFORE you install it & power the radio! Also by doing this you disconnect your headphone jack & it is no longer useable. 

[image]

poge-st, 2021-06-22:
thats some nice news - so in fact , what is happening is that the cables from the audio jack are disconnected and the "coupling" therefor is reduced ? 
maybe if i find such spare cable - ill really give it a try ! - any idea what this cable is / where to get a spare one ? :/

FLYN-001, 2021-06-22:
> thats some nice news - so in fact , what is happening is that the cables from the audio jack are disconnected and the "coupling" therefor is reduced ?
> maybe if i find such spare cable - ill really give it a try ! - any idea what this cable is / where to get a spare one ? :/

Yes exactly that! To go further you could try lifting the headphone L/R pins on the chip on the lower edge of the main board. 

Any good premium frsky dealer should carry spare parts including this ribbon cable. Check their website for your nearest supplier. 

I got mine from RCStudio (EU)

poge-st, 2021-06-22:
many thanks ! - ill have a look
thanks a lot !

brainbubblersbest, 2021-06-22:
Got mine from Horusrc US.
Maybe the edgetx guys would Code this for a test

FLYN-001, 2021-06-22:
> many thanks ! - ill have a look
> thanks a lot !

You’re most welcome, let us know how you get on!

FLYN-001, 2021-06-22:
> Got mine from Horusrc US.
> Maybe the edgetx guys would Code this for a test

That could be a good option, they seem to get things done pretty quick. ….touchscreen support 🧐

yuergen, 2022-08-22:
I did the modification to my Horus X12S. What worked for me:

- Use a scalpel to do the cutting of the ribbon cable.
- With a miniature flathead screwdriver you can feel the copper traces inside the cable and the dents between them when you slide it across the wire. Do just rest the flat side of the tip on the cable and move the tip across the cable.
- Rest the screwdriver after the 7th(!) copper trace. Cut between the 7th and 8th copper traces. Use the tip as a ruler by sliding the scalpel against the tip. A 1mm or 2mm cut will be enough.
- Now turn the screwdriver 180° and rest the tip between the 9th and 10th trace. Cut between these two traces, again using the tip of the screwdriver as a ruler.
- Add orthogonal cuts, to remove a section of the ribbon cable.
- Check with a multi-meter if you managed to interrupt trace 8 and 9 by dragging the tips on both side of the ribbon cable. The continuity check needs to indicate an open contact by not beeping when you check traces 8 and 9.
- Check with a multi-meter that you did **not** damage traces 7 and 10. Here you need to hear the beeping of the continuity check.
- Most importantly: Check with a multi-meter that you did **not** accidentally connect trace 7 and 8 **or** 9 and 10.

In addition to that I added copper foil to the lower housing. This might degrade RF performance because it is close to the antenna of the external module. I absolutely do not care about the internal module :). I guess the range of crossfire will be enough, even with some degradation, so this is not a problem for me. For you this might be different. I added Kapton / polyimide film above the copper foil, to prevent any short circuits. And I also added a grounding wire (the inductance because of the length and low cross section might make it ineffective, I did not test the performance without the grounding wire). Please see the attached pictures:

The buzzing is now nearly inaudible. In some situations it is inaudible/the radio is completely silent, as it should be from the factory.
[image]
[image]
[image]

FLYN-001, 2022-08-23:
> I did the modification to my Horus X12S. What worked for me:
> 
> * Use a scalpel to do the cutting of the ribbon cable.
> * With a miniature flathead screwdriver you can feel the copper traces inside the cable and the dents between them when you slide it across the wire. Do just rest the flat side of the tip on the cable and move the tip across the cable.
> * Rest the screwdriver after the 7th(!) copper trace. Cut between the 7th and 8th copper traces. Use the tip as a ruler by sliding the scalpel against the tip. A 1mm or 2mm cut will be enough.
> * Now turn the screwdriver 180° and rest the tip between the 9th and 10th trace. Cut between these two traces, again using the tip of the screwdriver as a ruler.
> * Add orthogonal cuts, to remove a section of the ribbon cable.
> * Check with a multi-meter if you managed to interrupt trace 8 and 9 by dragging the tips on both side of the ribbon cable. The continuity check needs to indicate an open contact by not beeping when you check traces 8 and 9.
> * Check with a multi-meter that you did **not** damage traces 7 and 10. Here you need to hear the beeping of the continuity check.
> * Most importantly: Check with a multi-meter that you did **not** accidentally connect trace 7 and 8 **or** 9 and 10.
> 
> In addition to that I added copper foil tape (self adhesive) to the lower housing. This might degrade RF performance because it is close to the antenna of the external module. I absolutely do not care about the internal module :). The range of crossfire will be enough, even with some degradation, so this is not a problem for me. For you this might be different. I added Kapton / polyimide film above the copper foil, to prevent any short circuits. And I also added a grounding wire (the inductance because of the length and low cross section might make it ineffective, I did not test the performance without the grounding wire). Please see the attached pictures:
> 
> [image] [image] [image]
> 
> The buzzing is now nearly inaudible. In some situations it really is inaudible/the radio is completely silent, as it should be from the factory.

Nice write up! Glad the mod worked for you. 
with the current CRSF firmware you’re able to program logical switches that automatically limit the tx power before & after an RX connection but keep it at full power in the event of a failsafe during flight. 

I wrote & use it myself, it works great! Total tx silence all the time!

AeroSeb, 2023-09-16:
Hi,

I have made the mod, but it doesn't work. The speaker always make loud sound and it is even worst with EdgeTX ! I cut the wire for headphones, move bluetooth cable etc... wihtout succes. It is very annoying. Hope another solution will work very soon.
Thanks for help.