opentx/opentx #8059: Choose which firmware version to download with Companion

jfrickmann, 2020-10-25:
A coupleof the latest releases have had serious bugs that have rendered my SoarOTX programs unusable. For people who have downloaded this firmware to their transmitters, there is no way to retrieve the previous version. This is a major problem for SoarOTX users, and we need a way to back out from buggy OTX releases. 

Therefore, I suggest that Companion gets a feature to select the firmware version to download.

3djc, 2020-10-25:
That would require multiplying our firmware compiling and distribution mechanism, this has serious cost (and time) implications.

jfrickmann, 2020-10-25:
In that case I will have to test new versions myself, and provide firmware files for the latest tested and approved version of OTX that works for my program.

3djc, 2020-10-25:
We are giving avanced warning on RCG that release is comming and asking poeple to test. if you want, we can warn you more specificalty that one is comming

lshems, 2020-10-25:
It really is a pity. It is not only relevant for soar otx users, but for any third party lua script provider.

With the current attitude where lua is considered a nice feature, but not more, and devs just want to have fun and do not want to take responsibility for the quality and user-friendliness of opentx, not much will change I am afraid. P.s. This is my synthesis of opentx dev statements. Please prove me wrong. 

Since I'm not taken seriously on this platform by the opentx devs, I doubt they will follow the following suggestion:

Have an emergency firmware available with a standard set of options and English only. Standard would be without any options ticked that only change layout (ppmus, ....) And enabling any options that activate extra functionality (lua, noheli off! No override off! Etc.)

This would also promote the use of these default option settings so that they can be made obsolete one day. They are the worst thing invented since beer in cans.

I have informed all users of my software, the paid justfly apps and free software downloadable on my site, NOT to use version 3.2.10 because of serious bugs.

I will start a repo as of now for those emergency firmwares. Might even ask money for them, as they are becoming precious.

Don't get me wrong, I do know the devs are entitled to do and don't anything they want with opentx. I just think it would be more fun for all if they would change their attitude nonetheless.

Have fun and just fly!

jfrickmann, 2020-10-25:
I assume the problem is that they "build to order" by re-linking firmware with the options selected in Companion, because there are so many combinations. When we get a working version of OpenTX again, I will simply get "vanilla" builds for all radio models and save them in my SoarOTX repository for my users. 
I have over 400 followers on my SoarOTX Facebook page, and many people on my RCGroups thread also. It just gets old having to tell them again and again that "yes you are SOL" and "no, there is nothing that I can do about it - wait for the next OTX update, hopefully they fix it".

3djc, 2020-10-25:
"no, there is nothing that I can do about it"
 
There is, test before release. We won('t rerlease if we know there is an issue

jfrickmann, 2020-10-25:
So what thread on RCG is it? This one?
https://www.rcgroups.com/forums/showthread.php?3395177-Official-OpenTX-version-2-3-Discussion-Thread/

3djc, 2020-10-25:
Yes, but as I proposed earlier, we won't mind contacting you directly before a planned release to give you a chance to test before release, just let us know where it is more convienient for you to be contacted

jfrickmann, 2020-10-25:
That would be great! How will you contact me?

3djc, 2020-10-25:
Thats what I'm asking :) Tell me how you would like to be contacted, and we will

jfrickmann, 2020-10-25:
OK, I sent you an email to the address provide on your GitHub profile. Please let me know if it worked!

3djc, 2020-10-25:
Got it :). I'll send you an email when we are nearing a release enough time in advance to give you a chance to test and feedback

Hobby4life, 2020-10-25:
As for most of things, make a backup of every bin file when a new release comes out.
So you can always go back and forth without need of companion or recompiling again.

jfrickmann, 2020-10-27:
Yes, I know. But many of the people who use my model templates for sailplanes don't know, and they are the ones that I am trying to help here.

3djc, 2020-10-27:
And we will do that together by ensuring that SoarOTX is tested with new code before it is released :)

closed 2020-12-04 by jfrickmann