opentx/opentx #8060: Updating AVR multiprotocol module firmware fails from Taranis X9D at openTX 2.3.10

jtaylor2, 2020-10-26:
openTX 2.3.10
Taranis X9D
AVR Multiprotocol Module

Updating AVR multiprotocol module firmware from Taranis X9D fails with error "Firmware update error - NoSync" at openTX 2.3.10.  This worked with openTX 2.3.9.

This was reported and discussed on RCG in the DIY Multiprotol Module thread in the following posts by myself and others:
https://www.rcgroups.com/forums/showpost.php?p=45700669&postcount=20192
https://www.rcgroups.com/forums/showpost.php?p=45704105&postcount=20196
https://www.rcgroups.com/forums/showpost.php?p=45692937&postcount=20188

As reported in the above links, flashing a stm module from X9D works and flash a avr module from TX16s works.  Flashing a avr module from X9D fails.  I believe the error is coming from multi_firmware_update.cpp line 247.

jtaylor2, 2020-10-26:
I used git bisect starting at 2.3.9 and ending at 2.3.10 and it points to 5bf197439a95338263007c501e9f382d68e278a8 is the first bad commit:
commit 5bf197439a95338263007c501e9f382d68e278a8
Author: Jakub <jklimasz@gmail.com>
Date:   Mon Jun 22 16:44:35 2020 +0200
    Support of FRM302 module. (#7533)
    AFHDS3 support added

```
2HP-Linux:~/frsky/opentx/git/opentx # git bisect good release/2.3.9
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad release/2.3.10
Bisecting: 139 revisions left to test after this (roughly 7 steps)
[511187fa76ae949c533dc21eab5302b2f62116e5] Update CREDITS.txt
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
Bisecting: 69 revisions left to test after this (roughly 6 steps)
[4e7f6e4f29754b1443dca351c8c5c79d098b15a7] Update CREDITS.txt
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
Bisecting: 34 revisions left to test after this (roughly 5 steps)
[77f4c18f4c55f007d91872c59a72f94a27b343fe] Frsky production script updated
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
Bisecting: 17 revisions left to test after this (roughly 4 steps)
[e4336857869a53e3fdc699c0ae333e46230995e8] Update CREDITS.txt
HP-Linux:~/frsky/opentx/git/opentx # git bisect good
Bisecting: 8 revisions left to test after this (roughly 3 steps)
[6d29faa48b269cb6150b4974707e27f655d4d462] Version increment (we probably forgot it a t 2.3.9 release)
HP-Linux:~/frsky/opentx/git/opentx # git bisect good
Bisecting: 4 revisions left to test after this (roughly 2 steps)
[2d01916055a8654f1ae44e3d574464b3de0c77cb] Preliminary T18 support (#7581)
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
Bisecting: 1 revision left to test after this (roughly 1 step)
[5d4f6679e5d4715223f7a8e8982d824de7c12594] Remove Avr boards (#7713)
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[5bf197439a95338263007c501e9f382d68e278a8] Support of FRM302 module. (#7533)
HP-Linux:~/frsky/opentx/git/opentx # git bisect bad
5bf197439a95338263007c501e9f382d68e278a8 is the first bad commit
commit 5bf197439a95338263007c501e9f382d68e278a8
Author: Jakub <jklimasz@gmail.com>
Date:   Mon Jun 22 16:44:35 2020 +0200

    Support of FRM302 module. (#7533)
    
    AFHDS3 support added

 companion/src/firmwares/CMakeLists.txt             |   1 +
 companion/src/firmwares/afhds3.cpp                 |  12 +
 companion/src/firmwares/afhds3.h                   |  12 +
 companion/src/firmwares/moduledata.cpp             |  41 +-
 companion/src/firmwares/moduledata.h               |   8 +-
 companion/src/firmwares/opentx/opentxeeprom.cpp    |  38 +-
 companion/src/firmwares/opentx/opentxeeprom.h      |   3 +
 companion/src/firmwares/opentx/opentxinterface.cpp |  47 +-
 companion/src/modeledit/setup.cpp                  |  51 +-
 companion/src/modeledit/setup.h                    |   1 +
 companion/src/modeledit/setup_module.ui            | 174 ++++-
 companion/src/modelprinter.cpp                     |   5 +
 radio/src/dataconstants.h                          |   3 +-
 radio/src/datastructs.h                            |  18 +
 radio/src/gui/128x64/model_setup.cpp               |  93 ++-
 radio/src/gui/212x64/model_setup.cpp               |  97 ++-
 radio/src/gui/480x272/model_setup.cpp              |  90 ++-
 radio/src/gui/common/stdlcd/model_setup_afhds3.cpp |  48 ++
 radio/src/gui/gui_common.cpp                       |  12 +
 radio/src/gui/gui_common.h                         |  26 +-
 radio/src/pulses/afhds3.cpp                        | 706 +++++++++++++++++++++                                                                         
 radio/src/pulses/afhds3.h                          | 595 +++++++++++++++++
 radio/src/pulses/modules_constants.h               |  10 +
 radio/src/pulses/modules_helpers.h                 |  64 +-
 radio/src/pulses/pulses.cpp                        |  77 +++
 radio/src/pulses/pulses.h                          |  38 +-
 radio/src/pulses/pulses_common.h                   |  21 +
[... 11 lines cut ...]
 radio/src/telemetry/telemetry.h                    |  11 +-
 radio/src/telemetry/telemetry_sensors.cpp          |   8 +-
 radio/src/translations.cpp                         |  15 +-
 radio/src/translations.h                           |  11 +
 radio/src/translations/cz.h.txt                    |   6 +
 radio/src/translations/de.h.txt                    |   6 +
 radio/src/translations/en.h.txt                    |   6 +
 radio/src/translations/es.h.txt                    |   6 +
 radio/src/translations/fi.h.txt                    |   6 +
 radio/src/translations/fr.h.txt                    |   6 +
 radio/src/translations/it.h.txt                    |   6 +
 radio/src/translations/nl.h.txt                    |   6 +
 radio/src/translations/pl.h.txt                    |   6 +
 radio/src/translations/pt.h.txt                    |   6 +
 radio/src/translations/se.h.txt                    |   6 +
 radio/src/translations/untranslated.h              |   9 +-
 radio/util/fwoptions.py                            |  30 +-
 55 files changed, 2435 insertions(+), 164 deletions(-)
 create mode 100644 companion/src/firmwares/afhds3.cpp
 create mode 100644 companion/src/firmwares/afhds3.h
 create mode 100644 radio/src/gui/common/stdlcd/model_setup_afhds3.cpp
 create mode 100644 radio/src/pulses/afhds3.cpp
 create mode 100644 radio/src/pulses/afhds3.h
HP-Linux:~/frsky/opentx/git/opentx # 
```

jtaylor2, 2020-10-26:
Looking further....the change that broke the avr update was changing /radio/src/targets/taranis/telemetry_driver.cpp line 126 probeTimeFromStartBit = 52 to 48.

I can do a pull request to put it back, but I don't know why it was changed in the first place and what putting it back might break.  @Jakub why was this changed?

```
diff --git a/radio/src/targets/taranis/telemetry_driver.cpp b/radio/src/targets/taranis/telemetry_driver.cpp
index f0068232b..488b8af4c 100644
--- a/radio/src/targets/taranis/telemetry_driver.cpp
+++ b/radio/src/targets/taranis/telemetry_driver.cpp
@@ -123,7 +123,7 @@ void telemetryPortInvertedInit(uint32_t baudrate)
       break;
     case 57600:
       bitLength = 35; //34 was used before - I prefer to use use 35 because of lower error
-      probeTimeFromStartBit = 52; //round down - 48 used in original implementation
+      probeTimeFromStartBit = 48; //round down - 48 used in original implementation
       break;
     default:
       bitLength = 2000000/baudrate; //because of 0,5 us  tick
```

jtaylor2, 2020-10-27:
@raphaelcoeffic it looks like you did the original implementation.  Can you comment on why you choose 48 for TELEMETRY_TIMER->ARR and your thoughts on why avr flash fails after the change to 52 and your opinion on putting it back to 48?

jtaylor2, 2020-11-01:
I am submitting a pull request to fix AVR multiprotocol module firmware updates by changing probeTimeFromStartBit form the changed value of 52 back to the original implementation of value of 48.  I haven't been able to find out why it was changed, but it broke AVR  firmware updates from the Taranis and it wasn't changed for the Horus which was left at 48.

3djc, 2020-11-01:
I understand that, but it can't be reverted without prior understanding of why it was changed

jtaylor2, 2020-11-01:
I understand.  The math says that 52 X .5us should be mid bit of the first data bit, which is why I figured he changed it.  But not knowing much about bit banging and soft serial I couldn't figure the latency of the interrupt and bit probe so couldn't really decide if that was correct.  And I also don't know why 48 (which works) was chosen for the original implementation.

However the change broke the avr firmware update, and the same change was not made to the Horus driver which is still 48 and still works so I figured I would submit a request to put it back on the Taranis and fix the avr flash.

closed 2020-11-07 by jtaylor2