opentx/opentx #8064: Frsky X9D+  , firmware 2.3.10, keep saying wrong SD card version files.

apexc, 2020-10-28:
Hi Open TX Team, 

First of all thank you very much for all the hard work. 
I really appreciate it a lot. 

I just flashed my X9D+ from 2.2 nightly built to 2.3.10. 
Everything went successfully even the bootloader. 

Downloaded the SD card content which is 2.3V0034. Replace all the files. 
But every time I boot the radio, it keep telling me that the version is wrong. 
Error prompt says it needs 2.3V0034. 
I have double and triple check the content and also the version. They are all good. 
Not sure what is wrong. Any help would be appreciated. 

PS : Not sure if the RTC error would play up against the SD Card issue as my button cell batt is dead.

3djc, 2020-10-28:
Your content is wrong, you can check it inside opentx.sdcard.version file which should be at the root of your SD. But please, this site is NOT for community support, only for bug tracking. Please go to community sites like rcgroups or openrcforums for community assistance

closed 2020-10-28 by 3djc

apexc, 2020-10-29:
Hi , 

Sorry if i posted to the wrong place. 
I am actually reporting the issue unless it is not an issue. 
The opentx.sdcard.version.txt is correct , I verified it. 
Its OK. I will head to community page for further troubleshooting for the moment.