opentx/opentx #8067: RadioMaster TX16S: There is no multirotor profile when creating a new model

IgorS102, 2020-10-29:
When adding/creating a new model, there is no profile for multirotor (quadrocopters) models, there is only a glider and an airplane to choose from, regardless of whether the noheli option is enabled or disabled when building the firmware in OpenTX Companion.

**General information**
 - OpenTX version: 2.3.9 and 2.3.10
 - Hardware: RadioMaster TX16S
 - OpenTX Companion 2.3.10 on Windows 10
[image]

3djc, 2020-10-29:
Wizard are left to the community to create, seems like no one has seemed to bother, maybe because no two heli are the same, and fbl heli can be setup like planes

closed 2020-10-29 by 3djc

IgorS102, 2020-10-29:
Did you decide to close because "most" are not concerned about it? This is called professional deformity. When an experienced person does not see the problems of an ordinary consumer. I'm new to this hobby and the lack of a pre-configured profile personally upsets me. A quadrocopter is not an airplane or a glider.

kilrah, 2020-10-29:
Every multirotor flight controller software needs different settings, and said settings depend on what's been set up on the FC itself. 
There's nothing a wizard can reliably automate.