opentx/opentx #8068: 2.3.10  Simulator not showing GVar value changes using Adjust and Source is Slider (TX16S Radio)

jmuxica, 2020-10-30:
**Parameters**: I am using a TX16S radio and have OTX Companion V2.3.10 installed, MacBook Pro, Catalina OS   
**Description**: When using a Special Function that loads a variable value (from a variable source on the Radio) into a Global Variable. When you open Simulator you expect to see the Global Variable display changing values as you move the input source (in this case a slider).
  
**To Reproduce**
Steps to reproduce the behavior:
1. create a Special Function with the following information:
SF1, ON, Adjust GV1, Source, LS, ON
2. Start the Simulator 
3. Display the monitor showing Global Variables
4. Move the Slider (LS in this example) and observe GV1 value (should change as you move LS)
5. It does not show any changes as I observed in my system

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

3djc, 2020-10-30:
Please supply OTX, just tested your exemple and it seems to be working fine

jmuxica, 2020-11-02:
Thx for the response.  Checked again and it seems to be fine now, and I cannot replicate it.  And I don’t have a root cause.   I will check again a few times and different OTX files, and if i cannot replicate then I will close the open issue.   Thanks

jmuxica, 2020-11-04:
Cannot be replicated.

closed 2020-11-04 by jmuxica