opentx/opentx #8073: Allow the usage of the UART6 on Horus Transmitters in the same way as on Radiomaster TX16S

brainbubblersbest, 2020-11-01:
I like to request the possibility for Companion.
To make a decission, if the internal BT is used or not.
So after a remove, solder Pads here for Uart 6 can be used for anything other. 
I know that in the Compiler is that Option. 
Maybe that can be done also for the GPS UART. 
Here is also a dedicated JST Connector existent. 

After almost a year Struggling with the Messy Horus BT,
A Real Serial BT Mirror would be a lot more usefull. 
Also the possibility of Telemetry in,  and the Trainer options like on the TX16S.

This would be an welcome alternative to the never ending Bluetooth Issue. 
Need a stabile solution to extract Crossfire and s.port Telemetry. Since the P13 and the x10 equivalent was never be that functional, this would be sort of a "Last Option" 
And a Big Enhancement in Connectivity.

Hobby4life, 2020-11-02:
+1 for a build option BT or Internal GPS

brainbubblersbest, 2020-11-02:
Yesterday i flashed opentx to a TX16S from a friend, without the bt option. I complete freaked out how easy it was, to get all working with two well documented hc05 Modules, configured at 115200baud.
Works a lot better with Serial Mirror, as with the factory installed Module.

Maybe the Options here can be Named 
GPSisSERIAL
and/or
BTisSERIAL

To avoid Users forget to choose that Option, as they Dont change anything by Hardware.

I also request this as a OTX User. Not as a Horus Customer. 
Just to make a statement 😉

brainbubblersbest, 2020-11-08:
Description of the pads
[image]

StefanKellerAC, 2020-11-09:
+1 for a build option BT or Internal GPS or UART or ? ...

ola-aberg, 2020-11-17:
+1 for a build option BT or Internal GPS or UART

vicewize, 2020-12-05:
+1 for build option

ramabt, 2020-12-10:
+1 for a build option BT or Internal GPS or UART

koakumaping, 2020-12-15:
+1 for a build option BT or Internal GPS or UART

wimalopaan, 2020-12-15:
+1 for a build Option for general UART (like in RM TX16s)

redjacketKR, 2020-12-25:
+1 for a build option BT or Internal GPS or UART

DIY420, 2021-01-27:
+1 for a build option BT or Internal GPS or UART
It would be really cool to have the pure serial functionality (without all the Bluetooth garbage) for telemetry out.
But on my Jumper T16 I found no way to get it (apart from taking the original CRFX stream@420kBaud, convert this by arduino to 115kBaud and send it to HC05).
What did you mean exaclty on Nov 2nd:  Options GPSisSERIAL and/or BTisSERIAL? What can I do with that hint? 
Is it only avilable when I make an own build (I'm not able to do yet...)

brainbubblersbest, 2021-01-29:
I came to that because i have the x12s with factory gps.
But I discovered the x10 dont use the uart4 at all. The hal.h Shows the pin is instead used for a alternate function.

GPS is Serial can be ignore since it bring that Feature just for the x12s.

Bt is Serial should be favorised. 

A way to change the uart on express Radios to the mentioned modes would be also genious. 

Why?

Because you can fly with internal HF AND same time connect a bt in the Jr bay. 

Or 
Utilize the RX and tx in module bay for a 3dr Radio.

kusnendar, 2021-04-21:
> Description of the pads
> [image]
> [image]

Do can change baud? Because my buildin BT can't show telemetry on app

brainbubblersbest, 2021-04-21:
Telemetry even with bt telemetry mirror is handled different on Horus based Radios then Taranis based ones.
Thats why no one with a  Horus based Radio can use it for the most Trackers with Horus BT connection.
First Thing thats different handled is the Baudrate. 
Dont know whats different  more, but the u360gts Project published a New Firmware with Frsky X CRC fix to get it working again. And thats also the Reason why i request This Feature.
This would make some Things easier in usage.

raphaelcoeffic, 2021-04-21:
Gentlemen, I won’t refrain you from publishing nice mods! Especially for the useless BT module on X12S.

That being said, I don’t think the team will work on it. But a PR would be definitely considered if provided.

kusnendar, 2021-04-22:
> Telemetry actually, with bt telemetry mirror is handled somehow different on Horus based Radios, then Taranis based ones.
> Thats why no one with a Horus based Radio can use it for the most Trackers with Horus BT connection.
> 
> First Thing thats different handled is the Baudrate. Yes.
> You cant change that!
> 
> Dont know whats different more, but the u360gts Project published a New Firmware with Frsky X CRC fix to get it working again. And thats also the Reason why i request This Feature.
> This would make some Things easier in usage.

How to change boud rate buildin BT?

brainbubblersbest, 2021-04-22:
> How to change boud rate buildin BT?

closed 2022-04-16 by stale[bot]