opentx/opentx #8078: TBS Nano RX telemetry - only 10 sensors

Pavelnew1, 2020-11-07:
I am using OpenTX 2.3.10  on Radiomaster TX16S. Crossfire mini TX is plugged and bound with Nano RX, both have latest stable firmware installed. I am doing Discover new sensors in Model setup -> Telemetry, and only 10 sensors are discovered. I've contacted TBS support, checked every possible setting with them, switched to another Nano RX - nothing helped and only 10 sensors are discovered. Can this be a bug in OpenTX? What other information is required from my side?
Many thanks in advance.

3djc, 2020-11-07:
Are those by any chance the crsf own sensors and you are missing the fc ones ?

Pavelnew1, 2020-11-07:
I have checked multiple set up instructions and manuals - even when connecting just TBS mini TX and Nano RX the number of sensors discovered should be at least 14. I have for sure connected FC as well (Matek F411), but still the number of sensors is 10 only. Tried various firmware on TBS (downgrading, upgrading again) - same result. That is the reason for my message here - trying to see whether OpenTX can cut sensor discovery somehow...  Otherwise, not even sure what to check...

3djc, 2020-11-07:
Well not quite. A CRSF receiver connecter to nothing brings 9 sensors. Could you put a picture of your discovered sensors ?

3djc, 2020-11-07:
[image]

Pavelnew1, 2020-11-07:
[image]

Pavelnew1, 2020-11-07:
I do have the same (+ one more ANT)

Pavelnew1, 2020-11-07:
However here is the link with 14 sensors: https://www.youtube.com/watch?v=0pihx42GtIs&ab_channel=RCVideoReviews

3djc, 2020-11-07:
Then there is no issue on our side, you have an issue with your FC telemetry connection, nothing OpenTX can help with I’m afraid

closed 2020-11-07 by 3djc

Pavelnew1, 2020-11-07:
And the problem, that when I solder Nano RX to FC, I get the same number of sensors (after I do "Discover new sensors"

3djc, 2020-11-07:
Look at the video, he has capa and more, that coming from fc

Pavelnew1, 2020-11-07:
OK, understood, thank you for your answers.