opentx/opentx #8079: Voz femenina OpexTX 2.3.10

Vidorlp, 2020-11-07:
Hola, ¿seria posible descargar la voz femenina que viene para la version 2.3.10? hay muchas traducciones que estan mal y me gustaria corregirlas para despues compartirlas.

Gracias.

3djc, 2020-11-07:
English only please

Vidorlp, 2020-11-07:
Hei, would it be possible to download the female voice in Spanish that comes for version 2.3.10? There are many translations that are wrong and I would like to correct them for later sharing.

Thank you.

3djc, 2020-11-07:
Have a look here : https://github.com/opentx/opentx/blob/2.3/radio/util/tts_es.py

It is computer generated out of this. Closing this as this is not an issue, but you can keep posting if you need our help

closed 2020-11-07 by 3djc