opentx/opentx #8081: Taranis QX7 Radio Setup

jinxster64, 2020-11-08:
I have updated my taranis qx7 with the right firmware and everything works right except for when i try to access the radio setup function by pressing and holding the page button. When i first turned on the Tx the first time i was able to access the radio setup but i made some changes in the setup and now i am unable to access the radio setup. I tried to reinstall the firmware and that did not work. is there another way to access the radio setup function besides pressing the page button and holding it for a couple seconds? I am at my wits end trying to figure this out and need help. Any suggestions will be greatly appreciated.
.

3djc, 2020-11-08:
There is nothing you can change in the settings that can prevent page long to work. The only thing I can think of is a hardware issue on the page switch that does not allow it to be maintained long enough

jinxster64, 2020-11-08:
So how would i fix that issue?

kilrah, 2020-11-08:
> except for when i try to access the radio setup function by pressing and holding the page button

Maybe becasue that is accessed by long MENU, not long PAGE?

closed 2020-11-08 by kilrah