opentx/opentx #8082: Backlight action no longer working (2.3.10)

rbak69, 2020-11-08:
In OpenTX and Companion 2.3.10 the backlight action is no longer working. I use this option on my Taranis to switch on the backlight on when SF is switched on. When upgrading from 2.3.9 to 2.3.10 I noticed this is now no longer working. Downgrading to 2.3.9 resolves the issue.

What I noticed is than in Companion 2.3.10 the action now seems to require a parameter (which I think is incorrect). Even using MAX as the paremater does not resolve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Global Functions" or "Special Functions" (both are impacted)
2. Add a new function that uses one of the Switches. Assign "backlight" as the action.
3. Save the model and write it to the transmitter or simply test it on the simulator

**Expected behavior**
When the switch is in the required position the backlight should be switched on.

Please note that in Companion 2.3.10 when you select the backlight action for a Global or Special Function it now expects a parameter and it requires you to enable the function. This was not requierd for OpenTX and Compamion 2.3.9 and lower.

**General information**
 - OpenTX version 2.3.10
 - Hardware: Frsky X9D
 - OpenTX Companion 2.3.10 on Windows 10

kilrah, 2020-11-09:
Fixed in nightlies. https://github.com/opentx/opentx/pull/8018

closed 2020-11-09 by kilrah

rbak69, 2020-11-09:
Thanks did not see this. Next time I will try the nightlies first!

rbak69, 2021-01-08:
I just installed the new version 2.3.11 today but the issue that was resolved in the 2.3.10 nighties is now still an issue in the new official release 2.3.11.