opentx/opentx #8087: FRSKY HORUS X10S XPRESS ACCESS (COMPLETELY BLACK SCREEN AFTER A MISTAKE INTRODUCING INCORRECT FILES...

Danielsvf, 2020-11-11:
Hello Sirs

Help, Help, Help

Can anyone help?

I have not back up from its original state...

Trying to upgrade my radio I mistaked introducing (I think in the flash memory) a incorrect file...

Now the radio is not putting on, neither the led of the switch all is like off...

My PC ( different computers tried ) does not detect the radio, just make the sound of something has been connected, but not appear the device.

Have anyone the way to make it run the bootloader and then the Flash memory? 

At the end following by 10 time the FRSky instruction manual creating a FIRMWARE folder in the micro SD I could make do the screen something as this attached, but after the second pic with only 3 lines of text the radio has been about 10 hours like that 
but not advance.

[image]
[image]
[image]

Thanks to everyone in advance.

3djc, 2020-11-11:
We are very sorry to hear what is happening to you, but this site on only to handle OpenTX bugs, which your situation obviously isn't. You should contact FrSky support or alternatively try on community supports sites like rcgroup or openrcforums

closed 2020-11-11 by 3djc

Danielsvf, 2020-11-11:
Ok, thanks