opentx/opentx #8089: FM sensor issue

RookFPV, 2020-11-12:
Taranis x9d+SE (old)   /   OTX 2.3.10
Tbs Crossfire micro v1/   3.78
TBS crossfire nanoRX /   3.78
Kakute F7v1.5             /   BF 4.2.4

in OpenTx 2.3.10 the FM sensor does not work correct - it will show “---” in My radio why ?
Works fine with earlier version of OTX, will it work from 2.3.11?

3djc, 2020-11-12:
There is no known issue around that, and since all the sensors are handled the same way, it seems unlikely that OpenTX is the cause of that

RookFPV, 2020-11-12:
TBS In his documentation he wrote "in OpenTx 2.3.10 the FM sensor does not work correct - it will show“ --- ”in your radio"
page 29 https://www.team-blacksheep.com/tbs-tracer-manual.pdf

3djc, 2020-11-12:
I have sent questions to TBS, because no one (including themselves) ever reported that I think

RookFPV, 2020-11-12:
I figured out when I wanted to use "INAV lua" which needs an FM sensor. And I know that OTX worked properly in the earlier version. PS Thanks for the quick response.

3djc, 2020-11-12:
I'm using iNav script with CRSF and 2.3.10 without issues

RookFPV, 2020-11-12:
I cannot use it properly due to a problem with the FM sensor after installing OTX 2.3.10

3djc, 2020-11-12:
I have asked details to TBS, lets see what they say tomorow

kamik2, 2020-11-13:
the note was written by me as i receiver many reports as mail and could create the behavior here. that's removed now

my testing setup (retested a few minute ago):

x9d+, colibri fc with BF 4.1.0, 
crossfire on 4.08 down to 4.00 and 3.78

- otx 2.3.10: fm == "---"
- otx 2.3.9: fm == "---"
- otx "tbs crsfshot version": fm == "sta" 

i also seen reports with frsky systems (r9 and xjt) on facebook with the exact same behavior.

3djc, 2020-11-13:
Yes, I'm talking to Remo, we will investigate

3djc, 2020-11-13:
The issue has been identified, a piece of code was deleted as part of another change. We need to understand why it was done at that time before a proper fix can be applied (but be assured that it will be fixed shortly)

closed 2020-11-14 by 3djc

machadofelipe, 2020-11-17:
Sorry guys, but it seems there is still something wrong, nightly 375 is always missing one letter for the flight modes. For instance, "OK" just shows "O" and "WAIT" shows "WAI".
@3djc I think you are cutting one digit out in the textLenght. Can you reopen the issue and check?

Thanks for the effort ;)

3djc, 2020-11-17:
Well according to CRSF section of BF I should have : 
[image]

and I have 
[image]
[image]
[image]

machadofelipe, 2020-11-17:
I was testing with iNav, but in BF it seems that when it is disarmed, it also should have an "*" at the end. Check line 290 at https://github.com/betaflight/betaflight/blob/94cd6504722c7e39eee5bd2e1a41507a03ca561d/src/main/telemetry/crsf.c (not sure myselft).

3djc, 2020-11-17:
Try tomorows nightly then

machadofelipe, 2020-11-18:
Thanks @3djc, seems to be perfect now ;)

3djc, 2020-11-18:
Sweet

PR #8093 (2020-11-14): Restore text sensors and prevent overflow
This fixes #8089