opentx/opentx #8091: Radioking tx18s suport

txemyprieto, 2020-11-13:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

pitersonfascist, 2020-11-26:
+1 vote

brainbubblersbest, 2020-12-12:
https://www.open-tx.org/newradios.html

d-hartmann, 2020-12-26:
Not all radios should get assigned to the latest branch... In general +1, but here, please complete the available radios and THAN assigne FR to new branch

sfjuocekr, 2021-01-11:
I flashed it with the T18 build and it works no problem, you just won\t be able to use debug analog to view the touchscreen position/cursor.

sfjuocekr, 2021-01-12:
So yea, instead of building it from the master 2.3 branch I downloaded the release "opentx-release-2.3.11.tar.gz" and extracted it.

I'm trying to build it like this:

cmake -DPCB=X10 -DPCBREV=T18 -DINTERNAL_MODULE_MULTI=YES -DDEFAULT_MODE=2 -DMODULE_PROTOCOL_FLEX=YES -DHARDWARE_TOUCH=YES -DUSB_CHARGER=YES -DLUA=YES ..

The expected end result after running make flash would be "welcome to opentx" after pressing power, but it stops before that happens and only shows the logo.

How do I properly build it for the specified target?

sfjuocekr, 2021-01-16:
I figured out what was wrong with my builds, I'll finish off the changes needed and add TX18S build support in the following days (if there are no objections against it).

kyon03, 2021-04-17:
nice

sfjuocekr, 2021-04-25:
I never continued working on it, I've tried to contact RadioKing as well to see if they were willing to release the sourcecode. Sadly, they have not responded adequately :(

For now I can recommend just using the Jumper T18 build, the radio is basically a T18 with added LCD controller and some changes to the way audio is played back. At least, I do not get the hissing sound with stock RadioKing firmware but on the T18 firmware I do get it. The hissing stops temporarily if you enter the global menu, but it does not bother me.

sfjuocekr, 2021-04-25:
The changes are up on my github page btw, if someone wants to mess about with it.

RipperGOT, 2021-09-26:
> The changes are up on my github page btw, if someone wants to mess about with it.

Hi!  What are the changes you made for?  Another question, do you have jitter problems?  
Check out this link. 
[https://github.com/EdgeTX/edgetx/issues/702]

ruedig, 2021-11-02:
When delivered, my Radioking TX18S showed the correct screen view for the boot loader and also for the model settings.
After the upload to OpenTX 2.3.14, the view of the bootloader screen is shown rotated by 180 °.

[image]

How can I change this ??
Flashing back to OpenTX 2.3.10 was unsuccessful.

Thanks for the help
Rudiger

sfjuocekr, 2021-11-02:
Use the T18 build, I can highly recommend you switch to EdgeTX.

anshchawla521, 2023-03-01:
Hi @sfjuocekr  did u manage to get the touch screen working on RK TX18S ?

sfjuocekr, 2023-03-07:
Yes, but I recommend you switch to EdgeTX and just flash the T18 build.