opentx/opentx #8096: curve editor bug (at uncommon curves)

FranzDurst, 2020-11-15:
When trying to create this curve
6 point (also with other nr of point), custom x
X: -100, +49, +50, +59, +60, +100          (Y doesn't matter)
or some other values of x, these values can not be set.
The problem seems to happen when some x values should be shifted quit a distance.
For example the original -60 x point can not be set above -20.
Defining this type of curve on Transmitter works.

3djc, 2020-11-15:
????
X needs to be monotonically increasing, so you x list is invalid.

Also please state version and enviroment used

kilrah, 2020-11-15:
Will be fixed in next nightly, in the meantime you can drag the points instead of typing values, that works.

closed 2020-11-15 by kilrah

FranzDurst, 2020-11-15:
Companion 2.3.10 on Windows 10
Wow, that was fast!
Thx Developers

commit d4f8bbb2 (2020-11-15): Limits for X spinbox values weren't updated when points were moved by typing values (fix #8096)