opentx/opentx #8102: CRC checking flaw in FrSky D16 causes short channel glitches

steffensol, 2020-11-17:
**General information**
 - OpenTX version (2.3.5 and above - unknown exact firmware versions, visible across multiple FW versions)
 - Hardware: Frsky X9D+ (Both 2019 and pre-2019 models. We are unsure which was used when bug discovered) & RX8R - D16 protocol. Bug has been seen once on a Horus X12S as well - but i don't have any specific logs to back that claim up.

*Introduction*

During flights when flying at long range, with the RC hand-controller - there is sometimes a glitch that can be seen on some of the RC channels. Unfortunately for us - this has happened on our killswitch channel - temporarily powering down motors in-air before turning on again shortly after.

The first case was seen in January 2020 - and it has appeared sporadically since, across multiple handcontroller (Taranis X9D series) and receivers(RX8R Pro) with unknown frequency (and unfortunately unknown exact FW version for the OpenTx controllers - but all have FW later than 2.3.5). None of our previous internal investigations have been fully conclusive as to what is the cause of the glitch.

The following screenshot shows a ~80ms glitch (incl rising/falling edge, main signal is active for 20-25ms), where the kill switch channel (Ch8 in OpenTX channel numbering from 1-8. Ch7 input when counting from 0) temporarily rises and then drops again. It is highly unlikely that this is caused by pilot error - as the other channels(apart from pitch, which is also glitching) are perfectly flat for multiple seconds before and after. Also the duration is so short that it should be physically impossible to move switches that fast.
[image]

The same can be seen here - but the duration of this has changed to around 1second. In this case only one channel glitches.
[image]

There has been some internal bug findings in the standard FrSky firmware that explains the same behaviour - where one or two channels temporarily glitch during poor RF conditions - caused by poor CRC checking mechanisms. 
The findings has been documented by a betaflight user on reddit - with use of sources; 
https://www.reddit.com/r/Multicopter/comments/g76xu7/frsky_protocols_v2xx_facts_and_discussion

To reproduce this bug: 
Unfortunately it is not easily reproducible - but it always happens when the RC controller is far away from the RC Receiver. E.g. 500m+ - Sporadically a glitch can be seen on one or two channels simultaneously - for either very short duration (100ms or less) - or for a full second.

It is not clear to us if this is specific to the FrSky firmware, or if this could also happen with the openTX firmware. Our findings suggest that it might be present here as well.

Is the OpenTX community aware of this issue, and has there been any tests/rectifications done in later FW? I cannot see any mentions of CRC checking in any release notes.

3djc, 2020-11-17:
Well, since it only happens at range, it means it is a RF issue, therefore not an OpenTX one (OpenTX does not deal in any way with RF communications)
So since this place only deals with OpenTX issues, probably best place to discuss this is rcgroups

closed 2020-11-17 by 3djc

kilrah, 2020-11-17:
>Is the OpenTX community aware of this issue, and has there been any tests/rectifications done in later FW? I cannot see any mentions of CRC checking in any release notes.

These CRCs are part of the module/receiver firmwares which OpenTX is completely unrelated to, we do not handle RF transmission in any way.

steffensol, 2020-11-17:
Thank you very much for clarifying this @kilrah @3djc .
Could you be so kind and point us in the right direction for handling receiver FW updates?

3djc, 2020-11-17:
Not sure if it will help, bu pt you might want to start here: https://www.rcgroups.com/forums/showthread.php?3507071-Important-Firmware-Update-%C2%96-ACCST-D16-2-1-x

kilrah, 2020-11-17:
The manufacturer of whatever RF system is being used.