opentx/opentx #8104: "Retrieve calib. and hw settings from profile" does not work  - Companion 2.3.10

sande005, 2020-11-18:
Companion 2.3.10
Taranis X9D+ profile

Read Models and Settings from Radio.
Go to "Edit Radio Settings" button.
"Store Calib. ...in profile" says that it saved.
Start new otx file.
Go to "Edit Radio Settings"
Hit "Retrieve calib." button.  No change to hardware or calibration pages.

Copy and Paste icons next to "Edit Radio Settings" do work when going from one otx to another.
Reading models from radio, then writing back (with or without changes to models) does preserve the hw and calib. settings, as would be expected.

kilrah, 2020-11-18:
Profile has been kind of deprecated for years, maybe it should go...

sande005, 2020-11-18:
It is nice to be able to tweak the Settings using Companion.
Perhaps, if the Radio Settings (Setup/Global/Trainer/Hardware) were a separate upload/editable/download option in the Profile. Maybe let calibrations only live on the radio.  Several groups of settings that can be applied to one Profile?  Then Models can be models only...
Users then have the option of working only on Models, leaving Radio Settings on their radios alone.  Or adjust just the Settings, from Companion, or via 2 step download, do both.
But, I have no idea how this would impact other types of radios....

kilrah, 2020-11-19:
You can always copy/paste radio settings from a file to the other, which is easier than dealing with the "settings in profile" thing.
This had some use back in the 9x days where radios were would often have widely different sets of complex hardware mods, and the feature was copying only settings that were not hardware dependent.

closed 2021-09-04 by stale[bot]