opentx/opentx #8105: [Horus] S.Port output only outputs polling frames

lukas-hosek, 2020-11-18:
I have a setup connected like this:

S.Port on the R9M module (or on the Horus itself, doesn't matter) -> inverter -> PC serial in

[image]

I'm only seeing polling frames (7E xx). I would expect full telemetry mirror.

Horus X10 Express running OpenTX 2.3.10
R9M 2019 module, 1.3.0 FLEX firmware

3djc, 2020-11-18:
Telemetry for ACCESS stuff is not using sport line, but is embedded in PXX2 lines. Not an OpenTX choice

closed 2020-11-18 by 3djc

brainbubblersbest, 2020-12-15:
So, this means opentx cannot mirror the telemetry that it used for showing in the telemetry screen?
This should be possible somehow. 
The telemetry Sensors discovered in telemetry setup cant be send out to the Serial?

And if using Mode telemetry instead of Mode telemetry mirror?