opentx/opentx #8113: OpenTX: Model Image file - name cannot consist of 6 characters (or more).

shauleiz, 2020-11-21:
**Bug description**
OpenTX: Model Image file - name cannot consist of 6 characters (or more).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a model image file, name it Acromstr.jpg, place it in the SD in folder IMAGES
2. From the radio (TX16s in my case) set this image as the model image. The image is displayed just fine.
3. Connect to PC (Win10) and start OpenTX Companion
4. Edit the model -> Setup tab -> Model Image
5. Image not displayed
6. Note: If I rename image file to am.jpg it is displayed

**Expected behavior**
If an Image is displayed on the Tx screen it should also be displayed on the OpenTX Companion.
If Name is invalid, an appropriate error message must be displayed

**General information**
FWVERSION  opentx-tx16s-2.3.9-otx (60e1edc1)
DATE       2020-06-14
TIME       16:58:11
req SD ver 2.3V0029
Companion     v2.3.10

TramALot, 2020-12-20:
i don't know if this helps but I get zero crashes if I use only images I create,
photo shop or ms paint seem to work equally well.

 before that it rewrote the background image in default theme folder all by it self

shauleiz, 2020-12-20:
I use my own images. I think the problem is with the limitation on the **file name**.
Right now I think it is limited to 5 characters + extension.