opentx/opentx #8119: TARANIS Q X7 ACCESS: SBUS Trainer Module Input misconfigured in hal.h?

bug400, 2020-11-23:
I did not succeed to read SBUS input from the external module heartbeat pin. Therefore I had a closer look at the current GitHub sources:

According to radio/src/targets/taranis/hal.h  the X7 ACCESS configuration parameters for the SBUS trainer input pin are:

#define INTMODULE_HEARTBEAT_GPIO                GPIOA
#define INTMODULE_HEARTBEAT_GPIO_PIN            GPIO_Pin_7  // PA.07
#define INTMODULE_HEARTBEAT_EXTI_PinSource      GPIO_PinSource7

#define TRAINER_MODULE_SBUS_GPIO_AF          GPIO_AF_USART6
#define TRAINER_MODULE_SBUS_USART            USART6
#define TRAINER_MODULE_SBUS_GPIO               INTMODULE_HEARTBEAT_GPIO
#define TRAINER_MODULE_SBUS_GPIO_PIN         INTMODULE_HEARTBEAT_GPIO_PIN
#define TRAINER_MODULE_CPPM_GPIO_PinSource INTMODULE_HEARTBEAT_EXTI_PinSource

In radio/src/targets/taranis/trainer (function init_trainer_module_sbus) the I/O port for SBUS trainer input is mapped to an USART as follows:

GPIO_PinAFConfig(TRAINER_MODULE_SBUS_GPIO, TRAINER_MODULE_SBUS_GPIO_PinSource, TRAINER_MODULE_SBUS_GPIO_AF);

which expands to:

GPIO_PinAFConfig(GPIOA,GPIO_PinSource7,GPIO_AF_USART6)

But according to the documentation for the STM32F407 there does not exist an alternate function mapping to an USART for pin 7 of Port A.

bug400, 2020-12-02:
I got the SBUS input working after patching the file radio/src/targets/taranis/hal.h in opentx 2.3.9:

 #elif defined(RADIO_X7ACCESS)
   #define INTMODULE_HEARTBEAT
   #define INTMODULE_HEARTBEAT_REUSE_INTERRUPT_ROTARY_ENCODER
-  #define INTMODULE_HEARTBEAT_RCC_AHB1Periph      RCC_AHB1Periph_GPIOA
-  #define INTMODULE_HEARTBEAT_GPIO                GPIOA
-  #define INTMODULE_HEARTBEAT_GPIO_PIN            GPIO_Pin_7  // PA.07
-  #define INTMODULE_HEARTBEAT_EXTI_PortSource     EXTI_PortSourceGPIOA
+  #define INTMODULE_HEARTBEAT_RCC_AHB1Periph      RCC_AHB1Periph_GPIOC
+  #define INTMODULE_HEARTBEAT_GPIO                GPIOC
+  #define INTMODULE_HEARTBEAT_GPIO_PIN            GPIO_Pin_7  // PC.07
+  #define INTMODULE_HEARTBEAT_EXTI_PortSource     EXTI_PortSourceGPIOC
   #define INTMODULE_HEARTBEAT_EXTI_PinSource      GPIO_PinSource7
   #define INTMODULE_HEARTBEAT_EXTI_LINE           EXTI_Line7
   #define INTMODULE_HEARTBEAT_EXTI_IRQn           EXTI9_5_IRQn
+  #define INTMODULE_HEARTBEAT_EXTI_IRQHandler     EXTI9_5_IRQHandler
 #elif defined(RADIO_X7) ...

The radio will not boot if the additional last definition of INTMODULE_HEARTBEAT_EXTI_IRQHandler is not present.

mariusvds, 2021-04-08:
I have the same problem, I would like to connect a ESP32 to the input pin of the Q X7S, but it is not working.
Apparently this is not (going to be) changed in newer versions... (Am I right?)
I also can not find a good PPM or CPPM library to test wether the input pin is working at all.

mariusvds, 2021-04-08:
I have the same problem.
I want to connect an ESS32 to the input pin of the radio.
As far as I understand it, it is not (going to be) changed in upcoming releases of openTX... (correct me when wrong)

closed 2022-04-16 by stale[bot]