opentx/opentx #8122: Missing or inconsistent wav files

SHezinger, 2020-11-24:
After changing the language of my radio I noticed that for example some english or german sounds are are missing.

76 files are only in english, 

47 file are only in german available

My Setup is:
Taranis X-Lite Pro
sdcard-128-64-2.3V0034
OpenTx 2.3.10

[image]

3djc, 2020-11-24:
Not an issue, those files are not required. If you want some more German added, then you can submit a PR

closed 2020-11-24 by 3djc

SHezinger, 2020-11-24:
But why is "engine on" required in geman and not in english?

3djc, 2020-11-24:
Only sounds in system are mandatory, all others are for convenience, many use third party sounds pack with hundreds of sound files

SHezinger, 2020-11-24:
Ok I got it. 
How can I add some files?
I understand that the wav files are generated.

Wouldn't it be nice, if all languages had the same tracks? It would be easier for users to follow a tutorial and "find" the track in their language. OpenTX could hold a constant list of track names from which the user could choose. If a specific file isn't available in the selected language the OpentTx could play the default language (english)

3djc, 2020-11-24:
Again, all the files _required_ are there and maintained in all languages. All the rest is user data.This is the file that needs to be modified for German: https://github.com/opentx/opentx/blob/2.3/radio/util/tts_de.py

SHezinger, 2020-12-01:
Sorry to bother you, but what ist the leading language or reference for all the system files?
I am working on the tts_de.py and compared the mandatory system files with the ones in tts_en.py.

I found out that these entries are missing in tts_en.py systemSounds:

             ("Fehler", "error"),
             ("Poti zentriert", "midpot"),
             ("20 Sekunden", "timer20"),
             ("30 Sekunden", "timer30"),
             ("Senderstrom zu hoch!", "highmah"),
             ("Sendertemperatur zu hoch!", "hightemp"),
             ("A1 schwach!", "a1_org"),
             ("A1 kritisch!", "a1_red"),
             ("A2 schwach!", "a2_org"),
             ("A2 kritisch!", "a2_red"),
             ("A3 schwach!", "a3_org"),
             ("A3 kritisch!", "a3_red"),
             ("A4 schwach!", "a4_org"),
             ("A4 kritisch!", "a4_red"),

I couldn't find entries like error, timer20 or a1_org in any source file.
They are only in:
tts_cz.py
tts_de.py
tts_es.py
tts_it.py
tts_pt.py
and in the sentencetables.cs of recorder and speaker.

So, I asume that they are leftovers from older version and could be removed, right?

kilrah, 2020-12-01:
Different people have been putting what they thought they wanted when they were maintaining their language, there's no "reference".

SHezinger, 2020-12-01:
These files are not used an can be removed in all .py files
             ("EEPROM formatiert", "eeformat"), 
             ("Poti zentriert", "midpot"), 
             ("20 Sekunden", "timer20"), 
             ("30 Sekunden", "timer30"), 
             ("A1 schwach!", "a1_org"),    
             ("A1 kritisch!", "a1_red"),   
             ("A2 schwach!", "a2_org"),    
             ("A2 kritisch!", "a2_red"),   
             ("A3 schwach!", "a3_org"),    
             ("A3 kritisch!", "a3_red"),   
             ("A4 schwach!", "a4_org"),    
             ("A4 kritisch!", "a4_red"),   

These two files are only used in PCBX9E
             ("Senderstrom zu hoch!", "highmah"),
             ("Sendertemperatur zu hoch!", "hightemp"),

This entry is not in the tts_en.py and I think could be removed from audio.cpp 
           ("Fehler", "error"),