opentx/opentx #8136: Virtual 6p Switch

SHezinger, 2020-12-03:
I tried to setup the momentary switches SE and SF of my new Taranis X-Lite Pro to increment or decrement the flight modes.
First with Logical Switches, Special Functions and GVars then I tried to implement it in a LUA script.
Both is very frustration.

I came up with the idea of implementing a virtual 6p Switch for every radio.
Start value is always 0 (no need to store it) and it can be incremented or decremented in the range from 0 to 6 with a Special Function like "Adjust GVx".

SHezinger, 2020-12-06:
I would like to programm it myself, but I am not sure in which sourcefile I should implement it.
Any sugestions?

Devil-IdT, 2020-12-18:
Hello, I like the idea. But please stick to six positions, as with the normal 6-position switch. So start with state 1 (default) and end with state 6. This is compatible with the hardware switches.
LG, Micha
sorry for my google-english.

alteman, 2021-01-13:
you need two logical switches
`L01: a<x GV1 6 AND SE`
`L02: a>x GV1 1 AND SF`
and two special functions
`SF1 L01 Adjust GV1 += 1`
`SF2 L02 Adjust GV1 -= 1`

That's it.

padcom, 2021-01-23:
A good option if you're using a flight controller (inav, betaflight, ardupilot) is to implement an N-position switch using trims. I, for example, use a 10 position switch to choose which parameter I'll be adjusting (VTx power, cruise speed, control smoothness, pitch and roll board adjustment, enabling autotrim or enabling autotune). You can do it using one global variable set to min -100 and max 100, two special functions that increment/decrement the value (let's say in increments of 25) and output that global variable on a port.

padcom, 2021-01-23:
I have that logic bound to aileron trim on channel 11 and then on channel 12 I just output the value of rudder trim to use as the +/- switch. It works like a charm!

closed 2022-04-17 by stale[bot]