opentx/opentx #8144: Jumper t18 booting issues

frizzle101101, 2020-12-08:
There seems to be an issue with booting the t18 that i have had nermous people on Facebook confirm there is an issue here.

 The issue seems to be when you boot up right after a shut down. Now the timing with when u press the power button seems to determine if theres an issue or not. I have shown the imidaite turn on after shutdown in this video where the whole transmitter gets stuck in black screen what i assume is a loop within the boot sequence because no button response the power must be cut. https://www.dropbox.com/s/k31048llc73t2nt/VID_20201207_224753.mp4?dl=0
Now sometimes when u turn on after having to pull the battery like this it will boot emergency mode where u just need to do a reboot, at this point in opentx the models are no longer being corrupted after entering this emergency mode.

The second manifest of this issue is the tx booting with a funny screen seems to be related but might not be(im not sure) because it seems to also be related to the timing which the user waits to repress the power button after powering off. this one however seems to also happen randomly even if you wait what seems to be a second or 2 befor turning it back on, i feel ive had this happen at the field when ive powered off tx to go get quad then when come back sit down and turn back on will do this. There maybe somthing happening as improper shutdown? no idea but it seems repeatable as shown in this video https://www.dropbox.com/s/k31048llc73t2nt/VID_20201207_224753.mp4?dl=0

It is harder to reproduce the second issue because the first issue seems to overshadowed it, the first issue was happening consistently with an imidaite repress of power after shutdown, the second issue seemed to show if u give half a second or a second between repress of the power button after shutdown

**General information**
 - OpenTX version 2.3.10
 - Hardware jumper t18
 - OpenTX Companion windows 2.3.10

raphaelcoeffic, 2021-06-07:
We solved this at EdgeTX. The issue seemed to be a alignment error caused by the BinAllocator. It has been fixed there. A PR has been made here as well.

frizzle101101, 2021-06-07:
Yay im glad someone figured it out

bluecoat97, 2022-02-16:
@raphaelcoeffic Hello sir!  I am hoping you can help me with a question regarding the T18 blank screen issue raised above.   I have a T18 running EdgeTX 2.6.0 and it is having the issue above.  Intermittent blacked out screen when rebooting quickly.  Do you have any other suggestions?