opentx/opentx #8145: crsf protocol rxnum warning

kreidler, 2020-12-08:
When changing CRSF protocol rxnum to anything different than 00 and respective rxnum is in use from e.g. D16 system shows up with the warning that rxnum is in use already.

Expected a behaviour limited to each module or protocol like decribed in https://github.com/opentx/opentx/issues/6786 

 - OpenTX version 2.3.10
 - Frsky X9D+

3djc, 2020-12-08:
While the requirement makes sense, on the radio itself, it is not quite possible without significant rewrite, as looking for the type of module/protocol used by another model would require loading the model, which is not possible

kreidler, 2020-12-20:
Thanks @3djc for the explanation.
When it is not possible to load a model (in advance) than I am wondering how the rxnum warning message can be generated over the different modules I use. I get the warning e.g. for the D16 rxnum 02 (internal module) as well as for the rxnum 02 on a 4in1 module in the same warning message together with the model name.
So, I assume that there must be already a table where the names and numbers are included which could be extended by protocol / internal / external.

3djc, 2020-12-20:
There is extremely limite table called header (https://github.com/opentx/opentx/blob/2.3/radio/src/datastructs.h#L499) that contains model name and model id, thats what is used