opentx/opentx #8148: LUA script automation test runtime and tools

witon, 2020-12-08:
When I develop LUA scripts for opentx, I'd like to use Test-driven development.  I do unit tests on the LUA modules those do not depend on the opentx LUA API by pure LUA.  To test the LUA scripts depend on the opentx LUA API, I run them in the simulator or transmitter. It's not convenient to automation test the LUA scripts in the simulator.

If the simulator provides some command-line options to run a script and provides API to control the simulator's switches, joysticks, telemetry data, etc, the LUA scripts can be tested by automation. It helps the software engineers developing more useful and high-quality opentx LUA to enhance the transmitter.

Another way to do the automation test is simulating clicks on the simulator UI and verifying by capturing log and screenshot.  It's hard to run in GitHub actions and unstable.

bsongis, 2020-12-09:
It's a good idea, perhaps you could give it a try ;)

witon, 2020-12-10:
> It's a good idea, perhaps you could give it a try ;)

Ok, I will try to learn QT's development and simulator's source and implement it. I hope it can work and dose not take a long time.

closed 2021-09-22 by stale[bot]