opentx/opentx #8149: INAV Lua script not indicating flight modes

Romulusg, 2020-12-08:
2.3.10  is not reporting FM status from the telemetry

2.3.9 worked perfectly

**Describe the bug**
when activating the inav lua script notifications of flight modes are not indicated.
Items like engines armed disarmed, ready to fly, angle, horizon, RTH are not indicated.  GPS status is reported.

**To Reproduce**
Steps to reproduce the behaviour:
0. Have an active Inav plane or quad bound ready
1. Start the inav lua script, wait for gps call outs
2. select arm and launch as normal
3. activate different flight modes and no call notifications from telemetry indicated

**Expected behavior**
FM and status should be indicated in the inav lua script.

This is a key Component when using dji as all modes are not reported in osd. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (2.3.10
 - Hardware (TX16s)

3djc, 2020-12-08:
What radio protocol ?

Romulusg, 2020-12-08:
Crossfire 

Todd

3djc, 2020-12-08:
Then it is fixed already in nightlies, and will be in 2.3.11

closed 2020-12-08 by 3djc

Romulusg, 2020-12-08:
Thanks. I read the nightlies and searched and didn’t see it. 

Todd

3djc, 2020-12-09:
https://github.com/opentx/opentx/commit/4de76302861c102cf3da5027d4acd2f077d94824