opentx/opentx #8155: Regression in FlySky telemetry reports bogus RSSI and RNse values

CapnBry, 2020-12-10:
**Describe the bug**
Receiving AFHDS2 telemetry items, the RSSI and RNse (noise) values are showing up as bogus positive numbers. Note: I am not referring to the rssi signal level values use by opentx to report signal levels, (which is RQly) this is the telemetry fields literally labeled RSSI and RNse which are informational only.

This was broken during the large AFHDS3 merge in pull #7533 by @bsongis. I'm not sure if this is because AFHDS3 reports its values differently, but this calculation is definitely wrong for AFHDS2. It might however be an oversight due to a bad merge because this calculation did used to be `value = 135 - value` but was fixed in #7042 by 3djc. It **was** 135-value because opentx used to use the RSSI value as the signal level source and it needed a positive value that went down the lower the signal is. We now use RQly as the opentx signal level value so fudging this number is not needed.

I'm a bit of an expert on these fields, having dug into the signal level / FlySky telemetry in the past and was quite pleased with how perfectly every telemetry field, including iNav telemetry, was represented in opentx compared an actual FlySky transmitter. The exception being RQly, which is inverted compared to the FS-i6 "Err." (error rate) field, but that's fine. I'm just reporting that these RSSI and RNse numbers are now back to being a somewhat meaningless scale.

**To Reproduce**
* Bind an AFHDS2 receiver.
* Go do the telemetry page, Discover sensors.
* RSSI and RNse are positive values in the range 30 to 75 instead of -60 to -105

**Expected behavior**
Comparing to an FS-i6 transmitter's telemetry screen, a maximum RSSI is -60dBm. OpenTX reports this as 75dB (135 - 60). Noise reported on my FS-i6 is -99dBm, and OpenTX reports 36dB (135 - 99). I would expect the values in OpenTX to be identical to what is reported by the hardware. What even is 75dB? -60dBm is an actual value. The sensitivity of the A7105 allows receiving down to -107dBm, although the FlySky 500kbig protocols are only resolvable down to -95dBm and the telemetry values are reporting directly on this scale. The telemetry value should be what the receiver is reporting back, not some arbitrary 135-X scale.

**General information**
 - OpenTX version 2.3.10
 - Taranis Q X7 (old version)
 - OpenTX Companion 2.3.9
-  Multiprotocol module iRangeX with Multiprotocol 1.3.1.78

kilrah, 2020-12-10:
> The telemetry value should be what the receiver is reporting back, not some arbitrary 135-X scale.

People wanted that from multi/opentx to have the same scaling for all protocols, so they're basically all "manipulated" to look like the original FrSky values.

CapnBry, 2020-12-10:
> People wanted that from multi/opentx to have the same scaling for all protocols and to be able to use the built-in warnings, so they're basically all "manipulated" to look like the original FrSky values.

But the built-in warnings don't run on RSSI or RNse, they run off of RQly-- a 100% down to 0% signal scale.

FrSky values also don't run 30-70 (I think a perfect signal is 85?) so having these values not match original hardware and also not map directly to FrSky confusing to everyone.

qba667, 2020-12-11:
@CapnBry @kilrah  I will take a look tomorrow but I can agree some unification is needed.

3djc, 2020-12-12:
The warning should not be an issue because like with several other protocols now, it can be done using rqly or an equivalent. I think it is time we get as close as possible to native telemetry when it is possible.

closed 2021-09-22 by stale[bot]