opentx/opentx #8159: X9D+ (before 2019) crashes when selecting trainermode "Master/Serial"

Tiefflieger68, 2020-12-11:
openTX 2.3.10 @ X9D+

if  "SBUS Trainer" is activated for serial port in hardware settings and selecting trainermode "Master/Serial" in model setup, the radio crashes and reboots in emergency mode.
Found the same description in issue #7027.
No matter if MPM is activated or not or if MPM is pluged or not.

Tiefflieger68, 2020-12-25:
is it fixed in die latest Nighties ?
The issue is still present in 2.3.11N400

closed 2021-01-04 by bsongis

3djc, 2021-01-04:
@Tiefflieger68 please test tomorow

Tiefflieger68, 2021-01-04:
will do tomorrow later the evening

3djc, 2021-01-04:
Thanks, please let us known both if it works or not 👍

Tiefflieger68, 2021-01-05:
my Test Equipment

Radio X9D+ @ openTX 2.3.11N405
Receiver XM (connected to serial port in battery bay)

Settings:
"SBUS Trainer" selected at Radio Settings

Tests:
1:
select "Master/Serial" (Lehrer/Serial in my radio cause of German language pack)
CHECK - no reboot of radio

2:
unplug/plug receiver from/to serial port
CHECK - lost and recovery of trainer signal detected

3:
SBUS signal is correctly decoded?
Defined a mixer with source "TR1"
CHECK - Stick moves of slave radio is visable in Mixer Monitor

Everything seems working well !

3djc, 2021-01-05:
Cool ! Thanks for checking that :) Will be in 2.3.11 that should be released in the coming few days

commit 2f334911 (2021-01-04): Merge pull request #8164 from opentx/bsongis/8159_reboot

Fixes #8159
commit a95df473 (2021-01-04): Fixes #8159
Still not tested, but it doesn't reboot anymore