opentx/opentx #8170: Problems with telemetry variable # 19 "FM" from INAV over TBS Crossfire

RomanFoltyn, 2020-12-20:
Subject:
Jumper T18; Radiomaster Tx16s and others.
INAV All versions
TBS Crossfire (all versions from 2.2 to the current 4.08)
Error **only  firmware generated via Companion** (all versions) -
Test with PC  WINDOWS 7 Windows 10

**Important Information!
When using the original companion firmware supplied with the transmitter or downloaded from the manufacturer Web, everything works correctly!**

Problem Description:
One of the most important parameters is information about flight mode of aircraft.

This information is sent as an ASCI string via variable # 19 with the name FM. If the firmware version supplied with the transmitter is working, it is displayed correctly and texts such as "WAIT" or "ACRO" or "MANUAL" are transmitted.

With firmware generated by **all versions of Companion**, this variable is incorrectly decoded and displayed in the Telemetry menu as ---. If the LUA is used, it is not displayed correctly and the message NOT OK appears.

This is a very important issue!

3djc, 2020-12-20:
Fixed already, use nightlies

closed 2020-12-20 by 3djc

RomanFoltyn, 2020-12-20:
Thank you for the answer!

From which version?
best regards

kilrah, 2020-12-20:
There's only one you can get, so...