opentx/opentx #8175: Bluetooth name not changing when using telemetry

craigywsm, 2020-12-21:
**Describe the bug**
Using x9d 2019 When using the Bluetooth feature and setting the name, does not change the broadcast name. When telemetry is selected

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Hardware
2. Click on bluetooth and set as telemetry 
3. Scroll down to name
4. When searching bluetooth shows up as hello

**Expected behavior**
Bluetooth should show as set name

**Screenshots**
[image]
[image]

**General information**
 - OpenTX version (2.3.10)
 - Hardware (Frsky X9D+ 2019 )

kilrah, 2020-12-21:
BT module issue, out of our control

closed 2020-12-21 by kilrah

craigywsm, 2020-12-21:
@kilrah please can you elaborate?

kilrah, 2020-12-21:
The firmware of the BT module isn't taking the name correctly.

craigywsm, 2020-12-22:
@kilrah Thanks, is this that OpenTX do not know the proper AT Command to set the name or the AT command is being ignored?
Either way how do we link this issue awaiting for a response from Frsky ?

kilrah, 2020-12-22:
They've known about it for a year but never addressed it.