opentx/opentx #8176: Wider vario deadband needed

gliderfly, 2020-12-22:
When using a vario sensor with a Taranis X9D the sink rate threshold seems too high. It gets really annoying hearing the sink tone for long periods. The lower end of the deadband should be at least -5ms. The lowest it can be set is -2.1ms. It would be great to only hear the sink tones only during the times the glider is actually in sinking air and not just the normal glide slope.

kilrah, 2020-12-22:
Umm if your glider is falling at 2.1m/s without being in sinking air it'a a brick...

gliderfly, 2020-12-22:
Yes that’s true but I think there must be something wrong with the vario calibration in the receiver.

3djc, 2020-12-22:
Then the fix should not be in OpenTX !

gliderfly, 2020-12-22:
Maybe so, but I can’t very well change that without replacing the receiver. A wider deadband would at least give better functionality even though the actual descent rate would not be accurate.

gliderfly, 2020-12-22:
Problem solved. I found there is an offset setting in the vario menu. Seems to be working better now.

closed 2020-12-22 by gliderfly