opentx/opentx #8180: Cant use Trm5 as Source at input when i use same Trm5 to Inc Dec GlobalVariabel at Spezailfunktion

hrenz, 2020-12-26:
openTx V2.3.10
at TX16S and Horus X10

I use Trm5 for source at input and then Input to mixer and Servo, all fine, all works when I move the Trm5,fine.  

Then I use Trm5 too at Spezialfunktion for Inc Dec the GlobalVariabel GV1
and Inc Dec GV1 works ok,
but since then Trm5 dont work as Input source, no change at Input5 when i move Trm5 

Then I change Trm5 to Trm6 at Inc/Dec for GV1 
and Trm5 at Inputs works again

Think there is an connection opened when I use Trm5 same time for changing GV

kilrah, 2020-12-26:
Increment/decrement keeps normal trim function, Adjust with trim as source doesn't by design.

closed 2020-12-26 by kilrah

hrenz, 2020-12-27:
Hy kilrah,
with TrimX as source I  can set exact values and Inc/Dec it in 1us or 2 us steps for a Sevotestprogram
I use the free Trm5 or Trm6 (or other trims) as source for Inputs/Mixer 
**and** want to store the Value of Trm5 or Trm6 in a GVx too for other use

With analog  I can do it, so i was surpriced I cant do it with no Trim switches

kilrah, 2020-12-27:
When you do "Adjust GVAR" with a trim as source then whenever the function is enabled the trim maps to a separate register, so that you can use one trim to adjust multiple GVARs to different values AND still use it as normal trim. 

Doing it this way doesn't:
[image]

>And its also not possible to use a GVx as Source for Input/Mixer

You use MAX source and GVAR as weight, same thing.