opentx/opentx #8181: Telemetry Lost and never come back

pablitous, 2020-12-27:
First of all i thught it was the lua telemetry plugin but its not. Because I have been looking OPENTX logs, and i ahve a blank of telemetry data from when I lost telemetry to when i restart the radio.

My old thoughts (bug report on luatelemetry when thought it was the plugin)
"Well, I already found someone that had the same proble, and I also had it with my preovious radio once. https://github.com/iNavFlight/LuaTelemetry/issues/395#issue-659823054

### Describe the bug
The problem is that i was flying I got Telemetry Lost warning voice and the radio start vibrating. When telemetry return in few seconds, alarms still waning me "telemetry lost" and radio continue vibrating.
I tried to reset telemetry in the radio while in flight, nothing happened.
I tried reset Flight in the radio while in flight, nothing happened. (well something worse happened, i lose control of my plane, but allarm still bugged)

Setup:
· Radiomaster TX16s
· R9M 2019 with ACCST last version on 915
· R9mm ACCST firm with last vesion also in 915
· INAV 2.6

### To reproduce

1. Fly
2. Lose Signal
3. Telemetry Lost Alarm starts sounding
4. Never stopped

[tx16smodels.zip](https://github.com/iNavFlight/LuaTelemetry/files/5744045/tx16smodels.zip)

 i can provide a log from the flight also."

3djc, 2020-12-27:
We have already heard similar story where r9 receiver stop sending telemetry, but while you say you had to restart radio, most say they had to restart receiver

pablitous, 2020-12-28:
> We have already heard similar story where r9 receiver stop sending telemetry, but while you say you had to restart radio, most say they had to restart receiver. At this point it seems common opinion is it is an module /receiver issue, not OpenTX

ok, if this happened to me again i will start the radio while in flight. And test again.
Im pretty sure i solve it restarting radio in the ground. But all happened so fast, will check again if i have the posibility.
So we can know if its r9 problem or opentx
Thank you anyway.

3djc, 2020-12-28:
Unfortunately it is not that simple. If turning there receiver off/fix it, we are sure it is not otx, however if it requires the radio off/on, then it could be either Otx or the module firmware

pablitous, 2020-12-28:
You are right, maybe i can try unpluging the r9m module while in the air first. and if it is not solved, i can try restart radio.
Thank you.

ampmam, 2021-02-02:
I have a similar issue on few occasions when using Fport protocol but have not encountered it when using SmartPort protocol.

closed 2022-04-17 by stale[bot]