opentx/opentx #8182: Some german textstrings for smal LCD(X9-Lite) and large LCD (X9D+) are to long at spezialfunktion

hrenz, 2020-12-27:
Hy found 
some textstings are to long at Spezialfunktion settings

could someone change it at german version please
for X9D (large LCD) String Telemetrie  into Telemetry oder Telem.
for X9Lite (smal LCD) String Knüppel into GSHQ

thanks

fabo137, 2021-03-19:
> for X9Lite (smal LCD) String Knüppel into GQHS

I don't think that adding "CQHS" is a good idea. CQHS is basically the german translation of "TAER". If a user has other settings than TEAR e.g. AETR, it might be confusing and wrong to put this translation string over here.
We can discuss about having "Sticks" instead of "Knüppel" because "Knüppel" in my opinion is quite colloquial when meaning a controlling mechanism like a joystick. The exact german translation would be "Steuerknüppel" which is even longer.

also changing the german term "Telemetrie" back to english is ratter a workaround than a real fix. 

I suggest to to close this issue, as the **correct** translation would result in **longer** **strings** and changing back a german word to english is not a real fix.

hrenz, 2021-03-20:
Hy,
GQHS 

ans Sticks are ok

Verschickt: Fr, 19. Mrz 2021 7:09

closed 2022-04-17 by stale[bot]