opentx/opentx #8188: Inputs and Trims Issue (Companion)

Alex00101, 2020-12-29:
**Describe the bug**
Apologies for a vague title, not sure how to describe the issue. Please open attached OTX file (zipped) and observe:

[trimtest.zip](https://github.com/opentx/opentx/files/5752519/trimtest.zip)

**To Reproduce**
Open attached OTX file in Companion, open the "Test Model" model. Please observe:
   -  Inputs: I1 (Ail) with TrimR (e.g. set for cross trim) and I7 (Nose). I7 uses rudder (nose steering) and has switch (F) assigned and Trim 6.
   - Mixes: Ch1 (Ail) uses I1 and Ch7 (nose steering) uses I7. 
 
Click "Simulate" and open Monitors.
   - Start clicking Trim R. Observe that **both**  Ch1 and Ch7 are affected. Expected: Only Ch1 should be affected.
   - Start clicking Trim 6. Observe that Ch7 is not affected. Expected: Ch7 should be affected. 
   - Either flick Switch F up or down or quit emulator, and remove Switch F from I7. In either case, now the model works as expected (Trim R affects Ch1 only and Trim 6 affects Ch 7 only).

**General information**
 - OpenTX version: 2.3.10
 - Hardware: Companion with X10 (internallaccess) profile
 - Windows 10

3djc, 2020-12-30:
Not seing this at all:

[image]

Alex00101, 2020-12-30:
OK, thanks. I was in a hurry removing unwanted entries (to produce clear model for review) and it got "corrupted". However you are correct, that beavior is consistent because I don't (and I didn't have) have the catcher line on I7. As soon as I added it, the model started to behave properly. Whether this "trim cross contamination" is a bug in the limited case when the catcher line is not defined as it was in my case - for developers to decide, please feel free to close if you think no further action is required.

Thank you.

3djc, 2020-12-30:
As a side comment, the same logic applies to your rate for exemple, where you use all 3POS position, you should define 2 and leave the third one as default

closed 2020-12-30 by 3djc

Alex00101, 2020-12-30:
Thanks, noted.