opentx/opentx #8190: [Improvement] Unique USB Vendor and Product IDs

netizen01, 2020-12-30:
A current issue with (almost) every OpenTX radio is that Windows changed how it ranks the appropriate driver for radios that are connected as HID Joysticks.

In September 2019, something changed in a Windows update to where the higher ranked driver became the "BETTER_USB_HS" driver published by STM. Exploring that INF shows that it matches the VID and PID exactly but does not classify the driver as a Joystick, thus it does not work as intended. The current workaround / solution is to switch the driver to the built in USB HID Device driver.

This workaround requires administrative privileges on the system and is a rather complicated task. For beginners who are just starting and learning to fly through a simulator, this is a huge hurdle to get over and many give up.

Certain radios are able to get around this issue by changing the Vendor ID to something other than STM, thus allowing the built in HID Driver to become the better match. The correct solution for each manufacturer (Taranis, RadioMaster, Jumper, TBS, etc.) would be for them to apply to the USB-IF and pay the $5,000 required to acquire their own Vendor ID. Unfortunately it seems none of them are willing to do that or have taken the time to investigate it. [TBS does have their own Vendor ID](https://devicehunt.com/view/type/usb/vendor/734C), but they do not actually use it no their Tango2 radio. Instead they switched it to [Microchip Technology's VID](https://devicehunt.com/search/type/usb/vendor/04D8/device/5710) to avoid this issue, though the Product ID is still the default).

Given the nature of this open source hardware / firmware, it seems an appropriate solution would be to find a way to obtain a VID / PID for any OpenTX radio. Understandably a $5,000 fee is not realistic. Fortunately there are a few solutions for OSS projects.

[https://pid.codes](https://pid.codes)

This site has acquired a Vendor ID from an older company before the USB-IF started prohibiting the transfer of VIDs. It is actively allowing OSS projects to use their VID and apply for a free unique PID through a simple Pull Request on their Github repository.

[I believe OpenTX qualifies.](https://pid.codes/howto/)

The process here would be to apply for a unique PID and then once approved, change the USB Vendor and Product IDs in the OpenTX Firmware.

**This fairly straightforward fix would solve the current windows driver issue that is present for all OpenTX radios** that do not create custom firmware images with a modified Vendor ID.

There are other options as well. STM will provide a unique Product ID that can be used in production if you use their chips in your products. I am unsure if they would give the Product ID to the OpenTX group though. Might be worth exploring as well.

Please consider this improvement as I believe it will dramatically improve the onboarding experience for new pilots who are wanting to use a simulator and may not be able to make the driver changes needed (school computer, not tech savvy, etc.).

i2bh, 2020-12-30:
Thank God someone finally fixed this!

netizen01, 2020-12-30:
> Thank God someone finally fixed this! The n00b killer glitch 😑 thnx netizen01

Not fixed yet, but this is at least a possible solution!

DownRightFPV, 2020-12-30:
Nice work!

batcavegames, 2020-12-30:
please look into this, the support that all sims have to go through daily fixing customers who don't know about this is a major hassle and a great annoyance to OpenTX users.

skyfpv, 2020-12-30:
Our team would greatly appreciate this fix being implemented as well! This has been the bane of our customer support efforts.

FPVanadium, 2020-12-30:
What he said. ^^^

lephro, 2020-12-31:
I agree. This is a huge pain and in an unexceptable state, for eveyone using openTX based products. You gotta fix this for at least 2.3 and the upcoming releases. FFS.

netizen01, 2020-12-31:
I've tested the change to the VID / PID on [this commit](https://github.com/netizen01/opentx/commit/79aa5ca32347b25783f1a8ac3fad771540945d0c).

Radio: FrSky XLite
Tested Joystick mode on Windows 10 and macOS - both worked as expected, Windows defaulting to the correct USB HID Driver instead of the BETTER_USB_HS (!).
Tested Mass Storage mode on Windows 10 and macOS - both worked as expected.

If the OpenTX team can request the suggested PID, I can open a PR on that commit and get this change merged!

mcw719, 2021-01-01:
Fixing this issue will have far-reaching effects. The pandemic triggered a boom in sim use, and our Youth Drone Sports Championships program was the only high school sport that didn't cancel last spring when we shifted to racing in VelociDrone. With virtually every student pilot needing to roll back their driver update to get their radio to work, we lost valuable time and many pilots bounced, meaning more time trying to bring them back. With this update, we can onboard many more students with far fewer hurdles. Thank you netizen01!

ZOETEK, 2021-01-01:
I agree with everything mcw719 said! cheers mate!! this is a urgent request!

juniordiscart, 2021-01-01:
As a Liftoff developer, we would love to finally see this solved. 95% of our controller related support questions are about them not being recognized due to this issue.

MooreMapping, 2021-01-03:
This would be a great fix to have! Really annoying to have to reset this every time there's a windows update. Thanks for the input Netizen01!

spinkham, 2021-01-06:
PR #8195 was merged into 2.3 branch recently with the [newly assigned ID](https://pid.codes/1209/4F54/).

closed 2021-01-06 by kilrah

bsongis, 2021-01-06:
Thanks for your help on this one @netizen01 

Thanks all other for the kind comments ;) please keep in mind that people helping here give their time on OpenTX for free, AFAIK OpenTX didn't get any free license / donation / whatever from simus companies mentioned in this issue.

ZOETEK, 2021-01-06:
Where could we send a donation? 

Kind Regards,

Jeff "ZOETEK" Grabowsky
SaltCtyRacingFPV, LLC
weBLEEDfpv.com
7256 LeafCrest Ln
East Syracuse, NY 13057

United We Fly, Together... weBLEEDfpv

batcavegames, 2021-01-07:
@bsongis perhaps you don't realise just how much this particular problem has cost the sim companies in support hours. Saying we didn't donate - we did - literally hours on end of support over the past year instead of redirecting all those people here for OpenTX support people to deal with. This problem has cost thousands in support time for us. I'll happily donate you some money if it makes you feel better but please try to understand this from our point of view - this has been one of the most costly problems we have ever had to deal with, the fact it is finally fixed is a blessing and we thank you for it but don't think for one moment we didn't pay for it.

kilrah, 2021-01-07:
> @bsongis perhaps you don't realise just how much this particular problem has cost the sim companies in support hours. Saying we didn't donate - we did - literally hours on end of support over the [...]

It's been a problem for you for a year yet this issue is literally the first time someone complained to us about it?

The problem was caused by a change out of our control, and it's the first time someone finds a way to fix it as well...

bsongis, 2021-01-07:
@ZOETEK here: https://www.open-tx.org/donations
@batcavegames well, no comment :)

DigoFuerte, 2023-05-22:
Please tell me there is a definitive fix for this problem - it is 2023 and this issue seems to have started in 2019. I just bought a Taranis X9D and LiftOff (to train myself to fly). My PC is Win 11 Pro, and the Taranis in Bootloader mode (v2.2.2) in OpenTX does not appear to connect, though I can see some of the start screen info from the X9D. LifOff reports that the X9D is not connected and in Win 11 Device Manager the only evidence of the X9D is in the Disk Driver branch where it appears twice as "FrSky Taranis USB Device". Any suggestions are welcome.

juniordiscart, 2023-05-22:
@DigoFuerte This is a whole different issue and the issue from OP was fixed in v2.3.11. If you're on an older version than that, you may still encounter the issue as described in the OP.

What you're experiencing is that your remote is set to function as a disk extension rather than a HID/gamepad. If you're still on v2.2.2, then perhaps the functionality is not present yet, but when connecting it to the system, it should provide you with a message on the radio to ask in what mode you want to connect it to the system.
If this is not the case, then there's still a setting somewhere that should toggle the radio's function upon connecting it to an external system. I would look into updating the firmware though, as that's imo the better course of action.

DigoFuerte, 2023-05-22:
Sorry, I can also see "FrSky Taranis Bootloader" in the Other Device section on the BlueTooth & Devices tab in Settings. It does not have any facility to update or change the driver... ?

juniordiscart, 2023-05-22:
@DigoFuerte updating is not done through device manager. Please check out the manual or support page of your radio on how to update it.

DigoFuerte, 2023-05-22:
@juniordiscart thanks for your response. I am a bit at this right because I am totally new to all of this... I am building a drone (piece by piece) and LiftOff was supposed to get me ready to fly when the drone is done. The Bootloader on the X9D is version 2.2.2 and on the FrySky support page that is the latest version. What am I missing there? Do I have to somehow install OpenTX on the X9D? 
When I start the bootloader, I see listed; Write Firmware, Restore EEPROM, Exit... then at the bottom
Curr FW: opentx-x9d+-2.2.2 (a5de713b)
Or plug in a USB cable for mass storage
Are any of these options what I need?

juniordiscart, 2023-05-22:
@DigoFuerte Those issues should be solved when performing a calibration of the controller in-game.