opentx/opentx #8191: Flaps SLOW UP and SLOW DOWN don't work in Crow-brake mix.

Helizero, 2021-01-01:
I did not find an existing bug-report about this.

Hardware: Radiomaster TX16S and Rx is Spektrum AR8010T

Software: OTX v2.3.10 
Radio: v. 1.3.1.92 AETR

2 servos for Ailerons (CH1 and CH5)
2 servos for Flaps (CH7 and CH8)

I am setting up a glider - Staufenbiel Mantis 2,9 meter with V-tail. Setting up Crow-mix so when i activate SG (second position)  the flaps goes down. When i flip SG to the back-position the flaps go further down and the ailerons go up (Crow brake) - this works fine but i can only get the Ailerons to move up and down slow like 1.5 seconds. Doing this with the flaps has no effect.

I am not sure if this is a bug or the mix is made wrong - but here is how it's done on the TX:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'GLOBAL VARIABLES' set GV1 to 30, GV2 to 40, GV3 to 45.
2. Go to 'MIXES' edit first line CH1 and under WEIGHT set it to -100% and do nothing else here.
3. Copy CH1 to make a new line.
4. Edit new line and under SOURCE choose switch SG and WEIGHT set to GV2 and SWITCH set to SG back-position. SLOW UP and SLOW DOWN set to 1.5. Back out to main MIXES screen.
5. Go to CH5 AilL and EDIT and set WEIGHT to -100% and do nothing else here. Back out to main MIXES screen.
6. Make a new line under CH5
7. Edit new line and under SOURCE choose switch SG and WEIGHT set to GV2 and set SWITCH to SG back-position. SLOW UP and SLOW DOWN set to 1.5. Back out to main MIXES screen.
8. CH6 not in use (Spare).
9. Go to CH7 (FlapsR) and EDIT and SOURCE is set to SG. WEIGHT is set to GV1 and OFFSET is set to -GV1. SWITCH is se to SG- (middle position) and SLOW UP and SLOW DOWN set to 1.5. Back out to main MIXES screen.
10. Make new line under CH7 (FlapsR).
11. EDIT new line and set SOURCE to SG and WEIGHT to 0% and OFFSET to -GV3 and SWITCH to SG in the back position. SLOW UP and SLOW DOWN set to 1.5. Back out ti main MIXES screen.
12. Go to CH8 (FlapsL) and EDIT set SOURCE to SG and WEIGHT to -GV1 and OFFSET to GV1. SWITCH is set to SG- (middle position). SLOW UP and SLOW DOWN is set to 1.5. Back out to main MIXES screen.
13. Make new line under CH8 (FlapsL).
14. EDIT new line set SOURCE to SG and WEIGHT to 0% and OFFSET to GV3 and SWITCH to SG in back position. SLOW UP and SLOW DOWN set to 1.5.
15. Back out to main model screen and test it.

I have added sounds to the SG switch in SPECIAL FUNCTIONS to play Track FlapUP, FlapHalf and CrowON.

**Expected behavior**
I was expected to see the Flaps go up and down at the set speed like 1.5.

regards

Kent

3djc, 2021-01-01:
This site is only for bug reports, not troubleshooting your model. Please go to community places like rcgroups or openrxforums to get you model fixed.

Also, supply an OTX, rather the require the person who wishes to help you to spend a lot time trying to redo your model instead of just opening it ...

closed 2021-01-01 by 3djc

Helizero, 2021-01-01:
Thank you for the reply.

I have talked to a youtuber that has the same issue and we therefore think
it might be a bug.

Also - i tried to attach the OTX file but the system does not support the
file-type.

regards

Helizero, 2021-01-01:
And just to enlighten you ..

When you start an issue it is clearly stated in the text-box what is
expected to be written.

Example:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

So i really do not think i have been treated fair by you - you could - just
to give some good service to a newcomer on this site - an explanation on
how to attach the OTX file instead of just fending off a user like that.
Still i thing the flaps should have been able to set to a certain speed of
deployment - right?

Den fre. 1. jan. 2021 kl. 11.44 skrev Kent Cortsen <wingnutz72@gmail.com>:

> Thank you for the reply.
>
> I have talked to a youtuber that has the same issue and we therefore think
> it might be a bug.
>
> Also - i tried to attach the OTX file but the system does not support the
> file-type.
>
> regards
>
>
>> This site is only for bug reports, not troubleshooting your model. Please
>> go to community places like rcgroups or openrxforums to get you model fixed.
>>
>> Also, supply an OTX, rather the require the person who wishes to help you
>> to spend a lot time trying to redo your model instead of just opening it ...
>>
>>
>
>
> Kent Cortsen

kilrah, 2021-01-01:
> When you start an issue it is clearly stated in the text-box what is
expected to be written.

And you supplied none of it.

For the OTX upload to dropbox, wetransfer or whatever...

> Still i thing the flaps should have been able to set to a certain speed of
deployment - right?

Sure and it's totally possible to do, so unlikely to be an OpenTX bug but improper configuration.

Helizero, 2021-01-01:
Okay and thank you for your time.

I have asked the question on RC Groups. Coming from the Spektrum system the
learning curve is a little steep on OTX - but still i love fiddling with it
- endless possibilities.

If i in any way seemed rude that was not my intention.

Happy new year to you - may 2021 be better than the past year.

Regards

-- 
Kent Cortsen